xreef/PCF8574_library #5: Design considerations

Pimmetje, 2018-06-01:
First of all thanks for this library. It used some macro's i haven't seen before.
I use library's like this on atmel 328. Those have 1024 bytes of ram. So i always try to limit the memory usage to as little as possible.

So i have some questions as i do not fully understand the bidirectional registry.
U can read and write only one byte to the register. If u write one byte u would read the same byte back. Because it would set a low powered pullup or down it's a in and output at the same time.

At startup all pins are high. Given no pins are grounded it will give 0xFF. 
So i only think u need two bytes or memory. One for (last) readed value. One for last written value. If i request the status i would be interested in the difference between the two. 
Maybe it could be useful to track if a pin was output or input. But on the other hand it does not matter for the device. So why spent a byte on it?

As for the status structure. Why use 8 bytes? 1 byte would do it?

It's not meant to criticize your work. I would like to learn from your considerations. And maybe mine can add something as well.

xreef, 2018-06-02:
The stats structure is a complex structure more flexible and direct usable, and not all like bitwise operation.
When I have more time I add a little enhancement to remove that structure with define flag for low memory usage.

The input and output mask is to prevent human error, and help readability of code.

Thanks

xreef, 2018-06-02:
I add the define variable, if you decomment PCF8574_LOW_MEM on lib you use byte instead of complex structure.

closed 2018-06-02 by xreef