xreef/PCF8574_library #7: digital read not working on PCF8574T

bmoniey, 2018-09-10:
Hello,

First of all, thank you for making this library available.

I purchased a couple IO expanders from amazon using the PCF8574T from phillips semi conductor.
I tried using this library to read the inputs, and could not return the correct result. I have 4 pins used as inputs and 4 as outputs driving LED's.

The datasheet states that pins used as inputs should be set to a high state.

In any case, I reworked your library and tested the inputs, which are working now. I also tested with using the interrupt pin.

I am using  teensy3.2 which is Arduino compatible, similar to an Arduino NANO.

I removed the buffered IO code, as it seems there really is no need for it (maybe on ESP8266). The interface to this chip is simply sending 1 byte for to set the outputs, and reading 1 byte to get all the ports.

Also, I think there should be some note somewhere that this device can't really be used to drive outputs high. The drive current is much less than 1mA when the state is HIGH. The user must rely on strong External Pullups to drive any HIGH loads, and would have to override the output by driving the pull-up and the load low.

I have attached my changes to the library code.

[PCF8574_cpp_h_changes_w_ex.zip](https://github.com/xreef/PCF8574_library/files/2365252/PCF8574_cpp_h_changes_w_ex.zip)

Thanks,

Brian

xreef, 2018-09-10:
Hi,
thanks for your help.

I read the code in a hurry, but I think there may be problems because you do not manage a buffer for readings, it may work with interrupts, but you would lose various readings in standard use.

The part not required, but inserted for future use is that of the masks specified in this very accurate analysis https://github.com/xreef/PCF8574_library/issues/5

However, I expected to enter the management of INPUT_PULLUP.

As soon as I have time, I add it.

Thanks again.

bmoniey, 2018-09-12:
Hello Renzo,

I was looking through the Arduino wire library code and it looks like for
Arduino, at least all the buffering of data is handled in the by wire.cpp
and twi.c. You can find there that all data buffer handling is done in that
Library.

Also, to note, the standard Wire library is blocking, meaning that reads
and writes to the I2C bus will wait until all bits are transferred from the
bus. which is why it should work inside the interrupts, even though it
caused the time inside the interrupt to extend for much longer.

The ISR example I gave was just so I could test out the attachement of the
interrupt to the class you made. Good Job!

Thanks again!

Brian

xreef, 2019-04-29:
I add support for second i2c channel, please check my site on usage
https://www.mischianti.org/it/2019/01/02/pcf8574-un-expander-i2c-i-o-digitale/

Thanks again for your support. Bye

Mark-Tec-Wizard, 2019-10-28:
Hello Xreef
I am using the library on an ESP8266.  As bmoniey  found in the above post the begin function and write function causes INPUT mode pins to write a 0 to the port pin.  This results in turning on the pull down transistor inside the PCF8574T.  When you pull up the pin externally, 32 ma is drawn to get the pin to go high.  This is not good for the chip if you have 8 inputs.  Also will drain unnecessarily drain the battery.
I made the following changes 
in the function begin
                //_wire->write(~usedPin);
		_wire->write(0xFF);
in the write function
             // writeByteBuffered = writeByteBuffered;   & writeMode;
              writeByteBuffered = writeByteBuffered;  

Mark

xreef, 2019-10-31:
Hi,
 I create new branch with a version that support input pullup, but not tested.
If you want you can try It first and please give me a feedback.
Thanks bye

Mark-Tec-Wizard, 2019-10-31:
Thank you Renzo I will try it over the next few days.
Mark -

alfredtorch, 2019-12-26:
Hi xreef,

I tried with no success. I defined everywhere INPUT_PULLUP, however it wasn't the solution to read buttons out. What I observed: changing P7 from Input_Pull to Input while keeping P6 untouched (as INPUT_PULLUP) I was able to get a feedback from that P6 Button, nothing on P7. Other combinations failed so far... Driving steppers and leds all going fine.

Thank you for your investigation.

xreef, 2019-12-28:
The code Is untested, I'm going to check It soon.
Thanks for your feedback Renzo

xreef, 2020-03-06:
Hi I finally write and tested the library, now on branch you can find the definitive version of the library.
I'm going to port It in the master repository when I update the tutoria.
Bye Renzo

closed 2020-03-06 by xreef