xreef/PCF8574_library #8: Using Arduino_STM32 Core

nopnop2002, 2018-09-14:
I tried with Arduino_STM32 Core.

I made alterations a little.

PCF8574.h

private:
	uint8_t _address;
//	uint8_t _sda = SDA;
//	uint8_t _scl = SCL;
	uint8_t _sda;
	uint8_t _scl;

PCF8574.cpp

void PCF8574::begin(){
	#ifndef __AVR
		#if defined(__STM32F1__)
			Wire.begin();
		#else
			Wire.begin(_sda, _scl);
		#endif
	#else
	//			Default pin for AVR some problem on software emulation
	//			#define SCL_PIN _scl
	// 			#define SDA_PIN _sda
		Wire.begin();
	#endif
 

Thank you for useful library.

xreef, 2018-09-21:
Please give me a feedback about integration of SMT32. 
Thanks

xreef, 2018-09-23:
I hope this fix..

xreef, 2018-09-24:
Thanks for your support and your patience. 
Bye

closed 2018-09-24 by xreef