xreef/PCF8574_library #10: Strange issue

RadioDJ, 2019-01-05:
Hi.
I have a strange behavier.
I have put all to output with led connected and it works, but when I start up my project all the leds are flashing 8 time with just a small light. Not full lightning.
This only happend when I put power to it the first time. If I disconnect the power and then put the power back it does the same thing.
All other things is working well.
Have I done something wrong??

xreef, 2019-01-07:
Hi,
can you put here the code and schema.
I start a personal blog, if you have troubleshooting to put here, you can try to put the data here https://www.mischianti.org/forum/forum/xreef-pcf8574_library/ .
Bye

RadioDJ, 2019-01-07:
The code I tested with is the BlinkOnPin.
The schema is that I found on your page.
I'm using a PCF8574P.

RadioDJ, 2019-01-07:
> The code I tested with is the BlinkOnPin.
> The schema is that I found on your page.
> I'm using a PCF8574P.

xreef, 2019-01-08:
Can you try with external power source please?

RadioDJ, 2019-01-09:
I have now tried with external power and it works much better. Only one flash is coming.
Before I tried with several computers, but all of them flash on all ports 8 times. With external only 1 time.
I have tried with only 1 port active but the same result. 8 times with usb and 1 time with external.

xreef, 2019-01-09:
Try add a 10uF capacitor to the power source, without external power source also. 
Keep in touch.

RadioDJ, 2019-01-09:
Already tried that. No difference. Same problem.
Tried with up to 220u.

xreef, 2019-01-11:
It's possibile that the IC don't have sufficient ampete to control all device, pcf have very low amperage, normally It's control trsnsistor.

RadioDJ, 2019-01-12:
In that case I wonder.
I have the same problem when I'm using only 1 led.
How do you initiate the pcf?
Do you initiate all the pins at once or one pin at the time?
Or is there anything else in the startup that can do this?

The only time the problem is shown, is at power up.
After everything is alright, I can light all the led without any problem.

xreef, 2019-01-14:
The pins are initialized all to low with one write.
Than start working with digitalWrite.

RadioDJ, 2019-01-21:
Hi.
I have checked it out put I can't figure out what it is.
I shall try some more things otherwise I must except this behavier.