xreef/PCF8574_library #11: PCF8575_library with Stepper.h library

ProRedCat, 2019-05-11:
Hello,

Thanks a lot for this library however, I'd like to use a PCF8574 chip to control some stepper motors. The problem I have encountered is when using the constructor of the stepper.h library, it's not exactly a problem with your library or the stepper library but I need to put the pins I'd like to use to control the stepper motor for example `Stepper myStepper(stepsPerRevolution, 8, 9, 10, 11);` where pins 8, 9, 10, 11 would instead be pins on the PCF8574 chip e.g. P0, P1, P2, P3 ect. However, with your library it appears you can only set a pin high or low and not parse them into the constructor for the stepper motor. Is there any way that this is possible or not or should I just not use the stepper.h library and write the code for this myself? Any help on this would be great. Thanks.

Regards,
ProRedCat.

xreef, 2019-05-11:
The stepper library use microcontroller pin not pcf8574 pins, but If I find an hour I fork the stepper library and modify It to use pcf8574.
Give me some time.

xreef, 2019-05-11:
I create a fork https://github.com/xreef/Stepper and I modify it to use pcf8574, check the example Testpcf8574Stepper.ino

Buti I don't know if It's work, pcf8574 have to low power, and stepper need a some power.

Tell me how It's work.

ProRedCat, 2019-05-11:
Alright, thanks a lot. I don't have a stepper motor on me right now so I'll test the code as soon as I can which should be about Monday. Once again thanks a lot for this.

Regards,
ProRedCat.

xreef, 2019-05-11:
I think to work you must use a MOSFET or transistor on any coil. But It's interesting thing, if I have some time I try.
Thank for your ideas

ProRedCat, 2019-05-11:
Hey,

The use case I have in this is I'm using four stepper motors which are each controlled by their own h-bridge. But doing it this way means I need 4 wires per h-bridge meaning 16 pins in total, so I've made some circuitry that makes it only use 2 wires per h-bridge but that still means I'm using 8 pins on the arduino, so what I decided to do is use a PCF8574 chip controlled by your library to control all the h-bridges. Because of this I never connect it directly to a coil as it's controlled by the h-bridge, so it should be fine. Thanks.

Regards,
ProRedCat.

xreef, 2019-09-25:
Inactivity

closed 2019-09-25 by xreef