xreef/PCF8574_library #12: Not work on nodemcu v3 PCF8575ST

masteriii, 2019-05-30:
I try to test on nodemcu v3  but not working..
/*
 Blink led on PIN0
 by Mischianti Renzo <http://www.mischianti.org>

 https://www.mischianti.org/2019/01/02/pcf8574-i2c-digital-i-o-expander-fast-easy-usage/
*/

#include "Arduino.h"
#include "PCF8574.h"

// Set i2c address
PCF8574 pcf8574(0x20);

void setup()
{
	Serial.begin(115200);

	// Set pinMode to OUTPUT
	pcf8574.pinMode(P0, OUTPUT);
  pcf8574.pinMode(P1, OUTPUT);
  pcf8574.pinMode(P2, OUTPUT);
  pcf8574.pinMode(P3, OUTPUT);
  pcf8574.pinMode(P4, OUTPUT);
  pcf8574.pinMode(P5, OUTPUT);
  pcf8574.pinMode(P6, OUTPUT);
  pcf8574.pinMode(P7, OUTPUT);
	pcf8574.begin();
}

void loop()
{
  int i;
  
	pcf8574.digitalWrite(P0, HIGH);
  pcf8574.digitalWrite(P1, HIGH);
  pcf8574.digitalWrite(P2, HIGH);
  pcf8574.digitalWrite(P3, HIGH);
  pcf8574.digitalWrite(P4, HIGH);
  pcf8574.digitalWrite(P5, HIGH);
  pcf8574.digitalWrite(P6, HIGH);
  pcf8574.digitalWrite(P7, HIGH);
  
	delay(1000);
  

  pcf8574.digitalWrite(P0, LOW);
  pcf8574.digitalWrite(P1, LOW);
  pcf8574.digitalWrite(P2, LOW);
  pcf8574.digitalWrite(P3, LOW);
  pcf8574.digitalWrite(P4, LOW);
  pcf8574.digitalWrite(P5, LOW);
  pcf8574.digitalWrite(P6, LOW);
  pcf8574.digitalWrite(P7, LOW);
  
	delay(1000);
}

xreef, 2019-05-30:
Hi masteriii,
I need more information to help you.
I work without problem on nodemcu.
Bye Renzo

masteriii, 2019-05-30:
Thank you  Xreef,
I use xreef/PCF8574_library  to control PCF8575 (IO Expander Module I2C To 16IO) to make 8 LEDs turn on/off but  anything happen with code above.   It can complie and upload to nodemcu v3 (ESP8266)

-------
PCF8575TS
https://www.aliexpress.com/item/PCF8575-IO-Expander-Module-I2C-To-16IO-Integrated-Circuits-New-2018/32862386560.html?spm=a2g13.10010108.1000001.11.4bec51f8WRTOGu

SP8266 CH340G NodeMcu V3
https://bit.ly/2JLrZTt

xreef, 2019-05-30:
Hi, this library is for pcf8574 not 8575.
Bye

closed 2019-05-30 by xreef

xreef, 2019-07-15:
https://github.com/xreef/PCF8575_library

I write a first release of a library for PCF8575, if you want try use the link upper.
Bye