xreef/PCF8574_library #13: Teensy 3.2 not compiling

dimitre, 2019-09-11:
It would be great to be able to use the library with Teensy 3.2 (and other teensies too)
Thank you

dimitre, 2019-09-11:
I can try to compile here installing teensyduino and selecting the board, even without the boards.
trying to compile any example I got this errors
```
/Users/z/Documents/Arduino/libraries/PCF8574_library/PCF8574.cpp: In member function 'void PCF8574::begin()':
/Users/z/Documents/Arduino/libraries/PCF8574_library/PCF8574.cpp:160:26: error: no matching function for call to 'TwoWire::begin(uint8_t&, uint8_t&)'
   _wire->begin(_sda, _scl);
                          ^
In file included from /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/Wire/Wire.h:29:0,
                 from /Users/z/Documents/Arduino/libraries/PCF8574_library/PCF8574.h:31,
                 from /Users/z/Documents/Arduino/libraries/PCF8574_library/PCF8574.cpp:28:
/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/Wire/WireKinetis.h:97:7: note: candidate: void TwoWire::begin()
  void begin();
       ^
/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/Wire/WireKinetis.h:97:7: note:   candidate expects 0 arguments, 2 provided
/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/Wire/WireKinetis.h:98:7: note: candidate: void TwoWire::begin(uint8_t)
  void begin(uint8_t address);
       ^
/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/Wire/WireKinetis.h:98:7: note:   candidate expects 1 argument, 2 provided
/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/Wire/WireKinetis.h:99:7: note: candidate: void TwoWire::begin(int)
  void begin(int address) {
       ^
/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/Wire/WireKinetis.h:99:7: note:   candidate expects 1 argument, 2 provided
Error compiling for board Teensy 3.2 / 3.1.
```

Thank you!

xreef, 2019-09-13:
Hi,
I add a fix,
but I use TEENSYDUINO key and I test It in Teensyduino
Please test this fix if It's correct.
Bye Renzo

dimitre, 2019-09-14:
@xreef thanks for the quick fix. I can compile and upload to Teensy. 
For some reason (probably my fault) I can't make it work with a simple setup I'm testing here (push button handling, with or without interrupt).
I have a protoboard and everything is working with Wire.h
Let me know if you want some additional testing

xreef, 2019-09-15:
Please, send me your working code with wire.
Thanks

xreef, 2019-10-14:
Close for inactivity

closed 2019-10-14 by xreef