xreef/PCF8574_library #15: PCF8574 output low active, init active

barneyz, 2020-02-14:
I use 8 relais modules with active low input, because the PCF8574 has active low outputs. On my init routine

`
	pcf_out1.begin();
	for (int i = 0; i < 8; i++)	// Init PCF
	{
		pcf_out1.pinMode(i, OUTPUT);	// 8 Ports init as output
		pcf_out1.digitalWrite(i, 1);
	}
`
With this code, all relais switch on at init for 100ms.

So i changed writeMode  and writeByteBuffered in the library from private to public and set both to a new init value:

`
	pcf_out1.writeMode = B11111111;
	pcf_out1.writeByteBuffered = B11111111;
	pcf_out1.begin();
	for (int i = 0; i < 8; i++)	// Init PCF
	{
		pcf_out1.pinMode(i, OUTPUT);	// 8 Ports init as output
		pcf_out1.digitalWrite(i, 1);
	}
`
This shortens the output init time to 10ms active time, so the relais are not able to switch. 

I do not understand the 10ms, a SSR will switch
Is it possible to init the PCF without any active time?

(I use an Wemos Lolin 32(ESP32) and #define PCF8574_LOW_MEMORY)

xreef, 2020-02-16:
Hi barneyz,
I must do some test, when I find sometime I try to manage It.
Thanks Bye Renzo

xreef, 2020-03-06:
Hi,
sorry to wait so much, i put on new branch a versione that support all

https://github.com/xreef/PCF8574_library/tree/INPUT_PULLPUP_Test

If you want try, now input are start LOW with INPUT, to have a HIGH value at first must be use INPUT_PULL_UP

 pcf8574.pinMode(P0, INPUT);
pcf8574.pinMode(P1, INPUT_PULLUP);

For OPUTPUT there is a new parameter and you can specify if start in LOW or HIGH

 pcf8574.pinMode(P6, OUTPUT, LOW);
 pcf8574.pinMode(P6, OUTPUT, HIGH);

I think this is risolutive.

Thanks

barneyz, 2020-03-07:
Thank you very much, i tested it and it works for me 👍

closed 2020-03-07 by barneyz

petr-novak-digibrocz, 2024-08-31:
Not working now, any changes?

Thank you