xreef/PCF8574_library #16: The digitalRead() does not work

kotakomputer, 2020-03-17:
I try digitalRead(P1) but does not work, result always LOW.
Using digitalReadAll(); the P1 result is HIGH (this is correct because I put 5V to P1).

kotakomputer, 2020-03-17:
SOLVED. Must using digitalRead(P1, true).

xreef, 2020-03-17:
Hi kotakomputer,
I think that It's usefully to undestand the problem.
How do you initialize the pin, like so
pcf8574.pinMode(P1, INPUT_PULLUP)
?

true in digital read bypass the debounce function that try to find a state change, so if you define P1 as INPUT_PULLUP and put HIGH the pin the state change not finded, but if you put only INPUT I think there is a problem in the library.

Please give me a feedback.
Thanks Renzo

closed 2020-04-02 by xreef

alirezaimi, 2020-05-15:
I have this problem,  and using true parameter in digitalRead method not work for me ,
result is always LOW .

  pcf8574.pinMode(P1, OUTPUT, LOW);
  pcf8574.pinMode(P2, OUTPUT, HIGH);

  pcf8574.digitalWrite(P1, HIGH);
  pcf8574.digitalWrite(P2, HIGH);
  int relay_1_state = pcf8574.digitalRead(P1, true); // LOW
  int relay_2_state = pcf8574.digitalRead(P2, true); // LOW

or 

  pcf8574.digitalWrite(P1, HIGH);
  pcf8574.digitalWrite(P2, HIGH);
  int relay_1_state = pcf8574.digitalRead(P1); // LOW
  int relay_2_state = pcf8574.digitalRead(P2); // LOW