xreef/PCF8574_library #23: pcf8574 and dht22

alirezaimi, 2020-06-05:
Hi
I want to read dht22 with pcf8574 and there is no luck in that until now !
I use adafruit library for dht, and this is my code :

	#include "DHT.h"
	#include "PCF8574.h"  // https://github.com/xreef/PCF8574_library

	#define DHTPIN P4     // Digital pin connected to the DHT sensor
	#define DHTTYPE DHT22   // DHT 22  (AM2302), AM2321

	DHT dht(DHTPIN, DHTTYPE);
	PCF8574 expander(0x20, 12, 13);  // esp8266 nodemcu and all Ax pins are Grounded

	void setup() {
	  Serial.begin(115200);
	  Serial.println(F("DHTxx test!"));

	  expander.pinMode(P4, INPUT_PULLUP);
	  expander.begin();

	  dht.begin();
	}

	void loop() {
	  // Wait a few seconds between measurements.
	  delay(2000);

	  // Reading temperature or humidity takes about 250 milliseconds!
	  // Sensor readings may also be up to 2 seconds 'old' (its a very slow sensor)
	  float h = dht.readHumidity();
	  // Read temperature as Celsius (the default)
	  float t = dht.readTemperature();
	  // Read temperature as Fahrenheit (isFahrenheit = true)
	  float f = dht.readTemperature(true);

	  // Check if any reads failed and exit early (to try again).
	  if (isnan(h) || isnan(t) || isnan(f)) {
	    Serial.println(F("Failed to read from DHT sensor!"));
	    return;
	  }

	  // Compute heat index in Fahrenheit (the default)
	  float hif = dht.computeHeatIndex(f, h);
	  // Compute heat index in Celsius (isFahreheit = false)
	  float hic = dht.computeHeatIndex(t, h, false);

	  Serial.print(F("Humidity: "));
	  Serial.print(h);
	  Serial.print(F("%  Temperature: "));
	  Serial.print(t);
	  Serial.print(F("°C "));
	  Serial.print(f);
	  Serial.print(F("°F  Heat index: "));
	  Serial.print(hic);
	  Serial.print(F("°C "));
	  Serial.print(hif);
	  Serial.println(F("°F"));
	}

xreef, 2020-06-05:
Hi alizeraimi,
the DHT library manage pin inside you can't mix the library without rewrite DHL lib.
Bye Renzo

closed 2020-06-05 by xreef

alirezaimi, 2020-06-06:
Thanks, can you explain for me what is DHL lib?

xreef, 2020-06-06:
Sorry DHT lib.

alirezaimi, 2020-06-06:
ok!
I edited dht lib, adafruit and troyka, but there is no luck in that :(
how can I use pcf with dht ?!

alirezaimi, 2020-06-06:
Is pcf8574 good for reading dht module ?
https://forum.arduino.cc/index.php?topic=505556.msg4630798#msg4630798

netpipe, 2021-01-24:
https://github.com/RobTillaart/DHTlib/issues/3