xreef/PCF8574_library #25: More than one expander

hmronline, 2020-07-21:
Could you please provide an example for initializing a few (let's say 4) gpio expanders ?
I was thinking on having an array, in order to be able to do something like this: gpioExpander[3].digitalWrite(7, HIGH);

Is that possible ?

Thanks in advance!

xreef, 2020-07-22:
Hi hmronline,
I think something like this can work

```
#include "Arduino.h"
#include "PCF8574.h"

PCF8574 pcf8574_1(0x38);
PCF8574 pcf8574_2(0x39);

PCF8574 arrayOfPCF[2];
//The setup function is called once at startup of the sketch
void setup()
{
	Serial.begin(115200);

	pcf8574_1.pinMode(P0, OUTPUT);
	pcf8574_1.pinMode(P1, OUTPUT);
	pcf8574_1.pinMode(P2, OUTPUT);

	pcf8574_2.pinMode(P1, OUTPUT);
	pcf8574_2.pinMode(P2, OUTPUT);
	pcf8574_2.pinMode(P3, OUTPUT);

	pcf8574_1.begin();
	pcf8574_2.begin();

	arrayOfPCF[0] = pcf8574_1;
	arrayOfPCF[1] = pcf8574_1;

}

void loop()
{
	arrayOfPCF[0].digitalWrite(P1, HIGH);
	arrayOfPCF[1].digitalWrite(P1, HIGH);
	delay(2000);
	arrayOfPCF[0].digitalWrite(P1, HIGH);
	arrayOfPCF[1].digitalWrite(P1, HIGH);
	delay(2000);

}
```

Give me a feedback.
Bye Renzo

hmronline, 2020-07-22:
Hi Renzo, thanks for you prompt response!

I've tried something similar before, but no luck with that, I've got an error on array declaration: no matching function for call to 'PCF8574::PCF8574()'

Any suggestions ?

xreef, 2020-07-22:
I try to compile my sketch and no error, can you post your code??

hmronline, 2020-07-22:
I'm using your code, but have to mention I'm testing it with platform.io (using visual studio code).
The provided error appears after hitting Build task.

Complete error message is:
```
src/main.cpp:7:21: error: no matching function for call to 'PCF8574::PCF8574()'
 PCF8574 arrayOfPCF[2];
                     ^
src/main.cpp:7:21: note: candidates are:
In file included from src/main.cpp:2:0:
.pio/libdeps/nodemcuv2/PCF8574 library/PCF8574.h:111:2: note: PCF8574::PCF8574(uint8_t, uint8_t, uint8_t, uint8_t, void (*)())
  PCF8574(uint8_t address, uint8_t sda, uint8_t scl, uint8_t interruptPin,  void (*interruptFunction)());
  ^
.pio/libdeps/nodemcuv2/PCF8574 library/PCF8574.h:111:2: note:   candidate expects 5 arguments, 0 provided
.pio/libdeps/nodemcuv2/PCF8574 library/PCF8574.h:110:2: note: PCF8574::PCF8574(uint8_t, uint8_t, uint8_t)
  PCF8574(uint8_t address, uint8_t sda, uint8_t scl);
  ^
.pio/libdeps/nodemcuv2/PCF8574 library/PCF8574.h:110:2: note:   candidate expects 3 arguments, 0 provided
.pio/libdeps/nodemcuv2/PCF8574 library/PCF8574.h:107:2: note: PCF8574::PCF8574(uint8_t, uint8_t, void (*)())
  PCF8574(uint8_t address, uint8_t interruptPin,  void (*interruptFunction)() );
  ^
.pio/libdeps/nodemcuv2/PCF8574 library/PCF8574.h:107:2: note:   candidate expects 3 arguments, 0 provided
.pio/libdeps/nodemcuv2/PCF8574 library/PCF8574.h:106:2: note: PCF8574::PCF8574(uint8_t)
  PCF8574(uint8_t address);
  ^
.pio/libdeps/nodemcuv2/PCF8574 library/PCF8574.h:106:2: note:   candidate expects 1 argument, 0 provided
.pio/libdeps/nodemcuv2/PCF8574 library/PCF8574.h:103:7: note: constexpr PCF8574::PCF8574(const PCF8574&)
 class PCF8574 {
       ^
.pio/libdeps/nodemcuv2/PCF8574 library/PCF8574.h:103:7: note:   candidate expects 1 argument, 0 provided
.pio/libdeps/nodemcuv2/PCF8574 library/PCF8574.h:103:7: note: constexpr PCF8574::PCF8574(PCF8574&&)
.pio/libdeps/nodemcuv2/PCF8574 library/PCF8574.h:103:7: note:   candidate expects 1 argument, 0 provided
*** [.pio/build/nodemcuv2/src/main.cpp.o] Error 1
```

hmronline, 2020-07-22:
Could you add an option to set I2C Address with begin function instead of setting it with class constructor ?

As an example, this library (which seems its abandoned) receives I2C address on begin function: https://github.com/microseti/pcf857x

It seems this way is easier to set an array of PCF8574 instances, something like this:

```
#include "Arduino.h"
#include "PCF8574.h"

PCF8574 arrayOfPCF[2];

//The setup function is called once at startup of the sketch
void setup()
{
	Serial.begin(115200);

       // This way, I2C addresses could be determined during execution time (maybe obtained with an i2cscanner function).

       arrayOfPCF[0].begin(0x38);
       arrayOfPCF[1].begin(0x39);

	arrayOfPCF[0].pinMode(P0, OUTPUT);
	arrayOfPCF[0].pinMode(P1, OUTPUT);
	arrayOfPCF[0].pinMode(P2, OUTPUT);

	arrayOfPCF[1].pinMode(P1, OUTPUT);
	arrayOfPCF[1].pinMode(P2, OUTPUT);
	arrayOfPCF[1].pinMode(P3, OUTPUT);

}

void loop()
{
	arrayOfPCF[0].digitalWrite(P1, HIGH);
	arrayOfPCF[1].digitalWrite(P1, HIGH);
	delay(2000);
	arrayOfPCF[0].digitalWrite(P1, HIGH);
	arrayOfPCF[1].digitalWrite(P1, HIGH);
	delay(2000);

}
```

What do you think ?

xreef, 2020-07-23:
Hi hmronline,
It's true, the sketch not working on Arduino, only in esp8266 core, try this
```
#include "Arduino.h"
#include "PCF8574.h"

PCF8574 pcf8574_1(0x38);
PCF8574 pcf8574_2(0x39);

PCF8574* arrayOfPCF[2];
//The setup function is called once at startup of the sketch
void setup()
{
	Serial.begin(115200);

	pcf8574_1.pinMode(P0, OUTPUT);
	pcf8574_1.pinMode(P1, OUTPUT);
	pcf8574_1.pinMode(P2, OUTPUT);

	pcf8574_2.pinMode(P1, OUTPUT);
	pcf8574_2.pinMode(P2, OUTPUT);
	pcf8574_2.pinMode(P3, OUTPUT);

	pcf8574_1.begin();
	pcf8574_2.begin();

	arrayOfPCF[0] = &pcf8574_1;
	arrayOfPCF[1] = &pcf8574_1;

}

void loop()
{
	arrayOfPCF[0]->digitalWrite(P1, HIGH);
	arrayOfPCF[1]->digitalWrite(P1, HIGH);
	delay(2000);
	arrayOfPCF[0]->digitalWrite(P1, HIGH);
	arrayOfPCF[1]->digitalWrite(P1, HIGH);
	delay(2000);

}
```

or 

```
#include "Arduino.h"
#include "PCF8574.h"

PCF8574 pcf8574_1(0x38);
PCF8574 pcf8574_2(0x39);

PCF8574 arrayOfPCF[2] = {pcf8574_1, pcf8574_2};
//The setup function is called once at startup of the sketch
void setup()
{
	Serial.begin(115200);

	pcf8574_1.pinMode(P0, OUTPUT);
	pcf8574_1.pinMode(P1, OUTPUT);
	pcf8574_1.pinMode(P2, OUTPUT);

	pcf8574_2.pinMode(P1, OUTPUT);
	pcf8574_2.pinMode(P2, OUTPUT);
	pcf8574_2.pinMode(P3, OUTPUT);

	pcf8574_1.begin();
	pcf8574_2.begin();

}

void loop()
{
	arrayOfPCF[0].digitalWrite(P1, HIGH);
	arrayOfPCF[1].digitalWrite(P1, HIGH);
	delay(2000);
	arrayOfPCF[0].digitalWrite(P1, HIGH);
	arrayOfPCF[1].digitalWrite(P1, HIGH);
	delay(2000);

}
```

Bye Renzo

hmronline, 2020-07-23:
Second option seems its working fine for me, thank you Renzo!

closed 2020-07-23 by hmronline

xreef, 2020-08-11:
Hi hmrontline,
I'm going to copy this thread on my forum site, i think It's interesting other people ask me the same example.
Bye Renzo