xreef/PCF8574_library #27: Can we use multiple digital pins of Arduino?

amandugar, 2020-10-12:
Can we use multiple Digital pins of Arduino like if we are using pin 2 to connect 8 pcf8574. Similarly, can we use 8pcf8574 with pin 3 together with pin 2 if yes please explain me how to do it

xreef, 2020-10-15:
Hi amandugar,
you can use multiple i2c channel to connect 8 pcf8574 each one by changing the address of pcf8574.
For esp32 you can configure TwoWire to do that.
More information here https://www.mischianti.org/category/my-libraries/pcf8574/ .
Bye Renzo

amandugar, 2020-10-16:
It's not that what i asked

xreef, 2020-10-16:
Sorry I don't understand, please try to explain better..

amandugar, 2020-10-16:
It's like we can use 1 digital pin of Arduino Uno to connect 8 PCF8574. As Arduino UNO has 14 Digital IO pins so can we connect more PCF8574? 
Or if we want more the 64 pins Can we Master-Slave I2c using Wire.h  or else how to do it if we want more then that

xreef, 2020-10-16:
Hi Amandugar, 
now I undestand, the i2c protocol need 2 pin, so you can't use only one pin to manage pcf8574.
Bye Renzo

closed 2020-10-16 by xreef