xreef/PCF8574_library #29: manage multi pcf574

carlosprato1, 2020-10-29:
Hello, very good bookstore, I am grateful. I would like to know if I can handle two pcf8474 at the same time? if yes. Do you have any example of how to do it?

carlosprato1, 2020-10-29:
recently testing I achieved it this way. excuse my little knowledge, thanks again.

`#include "Arduino.h"

#include "Arduino.h"
#include "PCF8574.h"  // https://github.com/xreef/PCF8574_library

TwoWire I2Ctwo = TwoWire(1);
TwoWire I2Ctwo0 = TwoWire(1);

 PCF8574 pcf8574(&I2Ctwo, 0x27, 21, 22);
 PCF8574 pcf85740(&I2Ctwo, 0x26, 21, 22);

void setup()
{
  Serial.begin(112560);

  I2Ctwo.begin(21,22,400000);
  I2Ctwo0.begin(21,22,400000); 
  delay(1000);

  // Set pinMode to OUTPUT
  for(int i=0;i<8;i++) {
    pcf8574.pinMode(i, OUTPUT);
    pcf85740.pinMode(i, OUTPUT);
  }

	Serial.print("Init pcf8574...");
	if (pcf8574.begin()){
		Serial.println("OK");
	}else{
		Serial.println("KO");
	}

  if (pcf85740.begin()){
    Serial.println("OK 0");
  }else{
    Serial.println("KO 0");
  }
}

void loop()
{
  static int pin = 0;
  
  pcf8574.digitalWrite(pin, HIGH);
  pcf85740.digitalWrite(pin, HIGH);
  delay(1500);
  pcf8574.digitalWrite(pin, LOW);
  pcf85740.digitalWrite(pin, LOW);
  delay(1500);

  
  pin++;
  if (pin > 7) pin = 0;
}`

xreef, 2020-10-30:
Hi Carlos, 
another similar issue on the forum.
https://www.mischianti.org/forums/topic/more-than-one-pcf8574-expander/
Bye Renzo

closed 2020-10-30 by xreef