xreef/PCF8574_library #30: OK / KO on initialization (code examples)

AlbertEmil, 2020-11-14:
I just started to use the library and stumbled upon the 

```
	Serial.print("Init pcf8574...");
	if (pcf8574.begin()){
		Serial.println("OK");
	}else{
		Serial.println("KO");
	}
```

used in many of the code examples.

Since this always prints "KO" for my circuit, I had a look in the library code and the Arduino documentation. Assuming that I did not miss something here and as far as I was able to investigate, the `pcf8574.begin()` function just references the default `Wire.begin()` function.

However, according to the [docs](https://www.arduino.cc/en/Reference/WireBegin), `Wire.begin()` returns `None`, so the if statement used in the examples will always evaluate to `False` and in turn state a `KO` status regardless of proper initialization.

xreef, 2020-11-15:
Hi AlbertEmil, 
I think you are not using my library, I check and I don't manage the status with wire.begin (It's a void function), I test It and the status work correctly with Arduino, esp8266 and esp32.
Please check and give me a feedback.
Thanks Renzo

AlbertEmil, 2020-11-15:
Hi Renzo,
I double checked the library I am using and it is definitely yours. As initially written, I always get a "KO" status upon initialization (when using one of your examples). Could you elaborate on the `pcf8574.begin()` function? Where is this boolean getting its value from? I just started with C and did not find any useful hint why I am always getting a "KO" instead of a "OK". However, the Port Expander is working just fine, even with a "KO" status.
Greetings, Al

xreef, 2020-11-16:
Mmm.. It's strange, can you give me more informations, the microcontroller, and a piece of your code.
Thanks Renzo

woyzeck, 2020-11-16:
I had the same issue today and from looking at the code of begin() my suspicion ist that if you don't set any pinMode before calling begin() it will fail.

xreef, 2020-11-16:
Hi woyzeck,
It's true I return a KO if pinMode isn't set.
Thanks for your support.
Bye Renzo

closed 2020-11-18 by xreef

dragosmacovei, 2024-01-10:
I have the same problem what do I have to do? it is unclear what solves the problem. pinMode is a function that applies to some specific pin. For what pin to call the pinMode function?