xreef/PCF8574_library #32: Reading from multiple encoders

poky, 2020-11-19:
First of all, want to thank you for creating this great library!
I'm trying to get two encoders to work, with following setups:

```
int encoderPinA = P0;
int encoderPinB = P1;
int encoderPinC = P2;
int encoderPinD = P3;
pcf8574.encoder(encoderPinA, encoderPinB);
pcf8574.encoder(encoderPinC, encoderPinD);
```

And read values by calling these functions:

```
pcf8574.readEncoderValue(encoderPinA, encoderPinB, &encoderValue1)
pcf8574.readEncoderValue(encoderPinC, encoderPinD, &encoderValue2)
```

However, the encoderValue1 and encoderValue2 can only goes up, no matter which direction I turn, but if I only one encoder, it works fine (clockwise +1 & counter -1)

Wondering if you could give some suggestions what I can implement this?

Cheers!

xreef, 2020-11-19:
Hi Poky,
I never test the internal implementation with 2 encoder, It's possible that there is a bug, 

to bypass the problem you can try to implement externally the management of 2 encoder, there is an example of external implementation on the relative tutorial

https://www.mischianti.org/2020/03/13/pcf8574-i2c-digital-i-o-expander-rotary-encoder-part-2/

But I try to fix as soon as possible.

Bye Renzo

xreef, 2020-11-21:
Hi poki,
change this lines
`
// Select an algorithm to manage encoder progression
#define BASIC_ENCODER_ALGORITHM
//#define MISCHIANTI_ENCODER_ALGORITHM
`

comment MISCHIANTI_ENCODER_ALGORITHM and decomment BASIC_ENCODER_ALGORITHM, and give me a feedback.

Thanks Renzo

poky, 2020-11-23:
> Hi poki,
> change this lines
> 
> ```
> // Select an algorithm to manage encoder progression
> 
> #define BASIC_ENCODER_ALGORITHM
> 
> //#define MISCHIANTI_ENCODER_ALGORITHM
> ```
> 
> comment MISCHIANTI_ENCODER_ALGORITHM and decomment BASIC_ENCODER_ALGORITHM, and give me a feedback.
> 
> Thanks Renzo

Hi Renzo, basic encoder algorithm is still doing the same (only increment when rotate)
However I get by this problem by modify the if statement as following:
```
if (na && !nb) {
					if (encoderPinBLast) *encoderValue = *encoderValue + 1;
					else *encoderValue = *encoderValue - 1;
					changed = true;
				}
```
Now both encoders are working correctly, thanks!

xreef, 2020-11-23:
Hi Poki,
thanks for your solution, but your solution not work with my encoder, the first encoder only increment when rotate, the second work correctly.
I think there are some difference from the encoder, so I'm going to add you r algorithm  to the library for all people that have your same type of encoder.
Now we have:
```
#define POKI_ENCODER_ALGORITHM
//#define BASIC_ENCODER_ALGORITHM
//#define MISCHIANTI_ENCODER_ALGORITHM

```

Thanks for your support.

I reopen this issue to show this behaivor to all people that watch this project.
And I'm going to add this solution to the support forum, so It has more visibility.

poky, 2020-11-23:
> Hi Poki,
> thanks for your solution, but your solution not work with my encoder, the first encoder only increment when rotate, the second work correctly.
> I think there are some difference from the [...]

Hi Renzo,

The strange thing is that if I only used a single encoder, the original algorithm works fine (01-10 clockwise, 01-11 counter)
But whenever I connected two encoders, the behavior changed to (00-10 counter)
Do have you a spare encoder to test when two connected? Thanks!

xreef, 2020-11-23:
Hi poki,
I had tested with 2 encoder and BASIC_ENCODER_ALGORITHM work fine for me, but with yours first encoder only increase.

But I'm going to to do more test.

Bye Renzo

xreef, 2020-11-23:
There is another bug 

`byte encoderValues = B00000000;`
must be
`volatile byte encoderValues = B00000000;`

Try It also please.
Thanks Renzo

poky, 2020-11-23:
> There is another bug
> 
> `byte encoderValues = B00000000;`
> must be
> `volatile byte encoderValues = B00000000;`
> 
> Try It also please.
> Thanks Renzo

Hi Renzo,

I've changed the PCF8574.h file as advised, but still the same

xreef, 2020-11-23:
Hi Poki, 
I checked and tested various encoder, and I have had multiple response different from different encoder.
So my solution to cover is this:

```
 #define BASIC_ENCODER_ALGORITHM
// #define MISCHIANTI_ENCODER_ALGORITHM
// #define SEQUENCE_ENCODER_ALGORITHM_REDUCED
// #define SEQUENCE_ENCODER_ALGORITHM
// #define POKI_ENCODER_ALGORITHM

```

I give the possibility to use different algorithm for different encoder like this
```
void updateEncoder(){
    changed2 = pcf8574.readEncoderValueSequenceReduced(P6, P5, &encoderValue2);
    changed = pcf8574.readEncoderValue(encoderPinA, encoderPinB, &encoderValue, true);

//  int vale = pcf8574.readEncoderValue(encoderPinA, encoderPinB);
//  if (vale!=0){
//      changed = true;
//  }
//  encoderValue = encoderValue + vale;

    bool val = pcf8574.digitalRead(P2);
    if (val!=valPrec){
      changed = true;
      valPrec = val;
      encoderButtonVal = val;
    }
    bool val2 = pcf8574.digitalRead(P4);
    if (val2!=valPrec2){
      changed2 = true;
      valPrec2 = val2;
      encoderButtonVal2 = val2;
    }
}

```

With this solution I can use different modules and different encoders with succes.

Give me a feedbback.
Bye Renzo

poky, 2020-11-24:
> readEncoderValue

Renzo, it works with the "reverseRotation" set true, but sometimes when turn clockwise, it will count -1

xreef, 2020-11-24:
Hi Poki,
if you have some exception on the behavior probably It's the noise, the encoders have a lot of noise.
Try to change the algorithm, or add some capacitor  to absorb It.
Bye Renzo

closed 2020-11-24 by xreef

xreef, 2021-01-03:
https://www.mischianti.org/forums/topic/pcf8574-reading-from-multiple-encoders/