xreef/PCF8574_library #33: PCF8575 integration

poky, 2020-12-07:
Hi, if I want to use both PCF8574 & PCF8575 in my project, what will be the best way do to?
It seems like PCF8574 library is the most up-to-date with encoder function supported, is there a way we can combine these two library into one? Or have to use them separately?

Thanks!

xreef, 2020-12-07:
Hi poki,
i spent more time on PCF8574 because I receive more feedback on It, pcf8575 seems It was few used.

If you want use this 2 device you must put the two different libraries on It.

If the pcf8575 will be used more, maybe in the future, I will bring it to the same level as the other library. With the little free time I have, I only focus on a few aspects.

Bye Renzo

poky, 2020-12-08:
Got it! Thanks!

closed 2020-12-08 by poky