xreef/PCF8574_library #38: Missing every other click of hw-040 rotary encoder.

QuitButton, 2021-01-30:
I'm having the problem of every other notch of the encoder getting lost, with the encoderValue only changing every 2 notch clicks when using a hw-040 rotary encoder.

This happens when using pcf8574.readEncoderValue(encoderPinA, encoderPinB, &encoderValue); and when using the example code within encoderWithBasicLibraryFunction example.

As a way of debugging I added 
`Serial.print("IRQ ");`
at the start of the void updateEncoder(). This seems to get called twice per notch, which seems wrong to me. 

Any ideas?

xreef, 2021-01-31:
Hi QuitButton,
I check that every type of encoder have a different behaivor, my solution is the implementation of many encoder alghoritm.

But I didn't update the documentatio, sorry, but It's quite difficult to follow all lib and pubblication (i need help if you are interested :P).

You can refer this topic 
https://www.mischianti.org/forums/topic/pcf8574-reading-from-multiple-encoders/

Give me a feedback about that, I try to update documentation as soon as possible.

Bye Renzo

QuitButton, 2021-01-31:
If I change to, say  #define MISCHIANTI_ENCODER_ALGORITHM I then get the error
'class PCF8574' has no member named 'readEncoderValue'

and changing the call to
changed = readEncoderValueMischianti(encoderPinA, encoderPinB, &encoderValue);
results in the error
error: 'readEncoderValueMischianti' was not declared in this scope

QuitButton, 2021-01-31:
I've not had any luck trying the other encoder algorithms, most of them give me no results at all.
However, going back to basics gives me this code, which for me is working nicely:

`volatile long encoderValue = 0;
volatile uint8_t encoderButtonVal = HIGH;
volatile uint8_t encoderPinALast = LOW;

void ICACHE_RAM_ATTR updateEncoder()
{
uint8_t n = pcf8574.digitalRead(encoderPinA, true);
uint8_t o=  pcf8574.digitalRead(encoderPinB, false);
if (n != encoderPinALast)
  {
  if (o != n) 
    {
    encoderValue++; 
    }
  else
    {
    encoderValue--;
    }
  }	
encoderPinALast = n;`

I don't know if the force parameter in digitalRead is needed or not, I was somewhat guessing part of the problem is the pcf8574 clears its interrupt on read.

xreef, 2021-01-31:
Hi QuitButton,
It's quite strange this code.
Force parameter is needed to bypass the internal library cache, so return the realtime value. If you don't use force if almost one time the value was go LOW It's return LOW.

I try to create a generic algorithm for all encoder, but every type give me a different result, I have 3 type of encoder and they work with the functions in the library.

For this type (20mm) 
https://s.click.aliexpress.com/e/_AZVINZ
of encoder with the schema in the article i use 
`    changed = pcf8574.readEncoderValue(encoderPinA, encoderPinB, &encoderValue, true);
`
for this type (blue but 20mm) with the schema in the article 
https://s.click.aliexpress.com/e/_A2tzDV
I can use the same function but not reversed
`	changed = pcf8574.readEncoderValue(ENCODER_PIN_A, ENCODER_PIN_B, &encoderValue);
`
for the same version but with the module 
https://s.click.aliexpress.com/e/_9vOthN
I must use 
`changed2 = pcf8574.readEncoderValueSequenceReduced(P6, P5, &encoderValue2);
`

If you find a generic function for your encoder that work with interrupt and force true I'm going to add to the function to manage in the library. Post also a link to your type of encoder.

Bye Renzo

xreef, 2021-02-04:
Close for inactivity!

closed 2021-02-04 by xreef