xreef/PCF8574_library #40: Outputs won't stay turned on for set amount of time

Creautre, 2021-03-08:
Hello, I am trying to control some track of a model railway.
I have two PCF8574 Modules connected to two 8-Channel Relay boards.

I am able to control the correct channels, but sometimes they won't turn on for the set amount of time of 700ms. They basically just bounce. I can barely see the LED Flashing on my Relay boards.

It happens to all channels on both boards at random. Sometimes it works flawless, most of the time it won't.

Here is a Video of my problem https://youtu.be/lHjU_x4Wy6U

This is my code

#include "Arduino.h"
#include "PCF8574.h"

// Set i2c address
PCF8574 Relais1(0x20);
PCF8574 Relais2(0x21);

const int Tschalt = 700; //Schaltdauer der einzelnen Elemente
const int gerade = 1;
const int umgelegt = 0;
const int geschlossen = 0;
const int offen = 1;
const int links = 0;
const int rechts = 2;

void setup() {
  Serial.begin(9600); //Aufbau Monitoring Verbindung
  delay(1000);
  for (int i = 0; i < 8; i++) { //Alle IO Extender als Outputs setzen
    Relais1.pinMode(i, OUTPUT);
    Relais2.pinMode(i, OUTPUT);
  }
  Serial.print("Init pcf8574...");
  if (Relais1.begin() && Relais2.begin()) { //Alle Relaisverbindungen prüfen
    Serial.println("Alle 3 Relais online");
    for (int i = 0; i<8; i++) //Alle Relais logisch High physikalisch Low setzen
    {
    Relais1.digitalWrite(i, HIGH);
    Relais2.digitalWrite(i, HIGH);
    }
  } else {
    Serial.println("Störung bei mindestens einem Relais"); //Störung an mindestens einem Relaisboard
  }
}

void loop() {
  Zug1();
  delay(3000);

  Signal(1,geschlossen);
  Zug2();
  delay(3000);

  Signal(2,geschlossen);
  Zug3();
  delay(3000);

  Signal(3,geschlossen);
}

void Zug1 () {
  Weiche(1,umgelegt);
  Weiche(2,umgelegt);
  Weiche(3,rechts);
  Signal(1,offen);
}

void Zug2() {
  Weiche(1,gerade);
  Weiche(2,umgelegt);
  Weiche(3,gerade);
  Signal(2,offen);
}

void Zug3() {
  Weiche(2,gerade);
  Weiche(3,rechts);
  Signal(3,offen);
}

void Weiche (int x, int y) { //Signalfunktion X ist die Signalnummer Y ist 
  if (x == 1) {
    if (y == 0) {
      Relais1.digitalWrite(0,LOW);
      Serial.println("Weiche 1 links");
      delay(Tschalt);
      Relais1.digitalWrite(0,HIGH);
    }

    if (y == 1) {
      Relais1.digitalWrite(1,LOW);
      Serial.println("Weiche 1 rechts");
      delay(Tschalt);
      Relais1.digitalWrite(1,HIGH);
    }
  }

  if (x == 2) {
    if (y == 0) {
      Relais1.digitalWrite(2,LOW);
      Serial.println("Weiche 2 gerade");
      delay(Tschalt);
      Relais1.digitalWrite(2,HIGH);
    }

    if (y == 1) {
      Relais1.digitalWrite(3,LOW);
      delay(Tschalt);
      Relais1.digitalWrite(3,HIGH);
    }
  }

  if (x == 3) {
    if (y == 0) {
      Relais2.digitalWrite(2,LOW);
      Relais2.digitalWrite(4,LOW);
      Serial.println("Weiche 3 links");
      delay(Tschalt);
      Relais2.digitalWrite(2,HIGH);
      Relais2.digitalWrite(4,HIGH);
    }

    if (y == 1) {
      Relais2.digitalWrite(2,LOW);
      Relais2.digitalWrite(5,LOW);
      Serial.println("Weiche 3 gerade");
      delay(Tschalt);
      Relais2.digitalWrite(2,HIGH);
      Relais2.digitalWrite(5,HIGH);
    }

    if (y == 2) {
      Relais2.digitalWrite(3,LOW);
      Relais2.digitalWrite(5,LOW);
      Serial.println("Weiche 3 rechts");
      delay(Tschalt);
      Relais2.digitalWrite(3,HIGH);
      Relais2.digitalWrite(5,HIGH);
    }
  }
}

void Signal (int x, int y) //Signalfunktion X ist die Signalnummer Y ist 
{
  if (x == 1) {
    if (y == 0) {
      Relais1.digitalWrite(4,LOW);
      delay(Tschalt);
      Relais1.digitalWrite(4,HIGH);
    }
    if (y == 1) {
      Relais1.digitalWrite(5,LOW);
      delay(Tschalt);
      Relais1.digitalWrite(5,HIGH);
    }
  }

  if (x == 2) {
    if (y == 0) {
      Relais1.digitalWrite(6,LOW);
      delay(Tschalt);
      Relais1.digitalWrite(6,HIGH);
    }
    if (y == 1) {
      Relais1.digitalWrite(7,LOW);
      delay(Tschalt);
      Relais1.digitalWrite(7,HIGH);
    }
  }

  if (x == 3) {
    if (y == 0) {
      Relais2.digitalWrite(0,LOW);
      delay(Tschalt);
      Relais2.digitalWrite(0,HIGH);
    }
    if (y == 1) {
      Relais2.digitalWrite(1,LOW);
      delay(Tschalt);
      Relais2.digitalWrite(1,HIGH);
    }
  }
}

//  vim:sw=2:ts=2:et:nowrap:ft=cpp:

xreef, 2021-03-08:
Hi Creautre,
the relay needs a good amount of power, is it possible that there is not enough current?
Bye Renzo

Creautre, 2021-03-08:
Forgot to add that information. I am using a external Labbench Powersupply set to 5.2V. Currentdraw (as I mostly only switch 2 relays at once) as around 200mA. Voltage never seems to sag judged by the old analog display and the display of my multimeter.

xreef, 2021-03-08:
Is possible that you want to hold the relay active with signal HIGH of pcf8574 probably you have some trouble because pcf8574 have very low power, less than a normal micro-controller. 
So you must work with LOW signal on activation if the relay board support this mode. Or you must use a transistor like 2n2222.

Can you give me some wiring information?
Bye Renzo

Creautre, 2021-03-08:
Here is a picture. It's directly soldered and the relays switch when they Relayboard inputs are pulled down by the PCF9574.
In my code you can see I switch the relays by sending a LOW to the ouput
```

if (y == 0) {
      Relais1.digitalWrite(4,LOW);
      delay(Tschalt);
      Relais1.digitalWrite(4,HIGH);
    }
```

The external Labbench Powersupply connects to the "end" of the I2C bus, injecting the 5.2V there.
This powers the PCF Modules, the relay boards and the arduino via Vin pin. 
There is nothing special going on. Just the I2C connection, VCC and GND. No caps, nothing.

[image]

[image]

xreef, 2021-03-08:
I don't understand what the problem could be, but in these cases it is almost always a power problem.
Is it possible that the cables are too thin? they seem to me like the jumper wires used for breadboards, they are not very efficient.
Can you put a thicker wire on the relay power supply?
Hi Renzo

Creautre, 2021-03-08:
I don't understand the issue either. Defies all logic to me. I will use thicker gauge next time I get to it. The setup is not at my place right now.

Do you still have some ideas or things I could try other than the powersupply changes?

xreef, 2021-03-08:
Try to check with multimeter the output of pcf8574, or better If you have more pcf8574 try to test the output with multimeter without connect to relay board.
Bye Renzo

Creautre, 2021-03-08:
Will do

Still kinda odd seeing this guy have no issues with 8 boards. I mean he has a beefy USB Powerbank, but the gauge on those USB Cables is not the fattest. 

But yeah he might just be lucky through the shorter cable. If I notice stability increase I will probably add so caps to each board be 100% sure the power is stable

Creautre, 2021-03-09:
The issue were those damn cheap PCF boards. They only have pullup resistors on them and nothing else.

I added an electrolytic 10uF cap and a 100nF cap. Now it's running fine. Sorry to bother u.

closed 2021-03-09 by Creautre

xreef, 2021-03-09:
Hehehe.. don't worry apicales. But can you contact me? Do you like to write an article on your train project?
Bye Renzo