xreef/PCF8574_library #41: Incorrect value from digitalRead() after startup

jhengstermann, 2021-03-16:
At power on, the I/Os are default High ([as shown in datasheet)](https://www.ti.com/lit/ds/symlink/pcf8574.pdf?ts=1615887401233&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FPCF8574%253Futm_source%253Dgoogle%2526utm_medium%253Dcpc%2526utm_campaign%253Dasc-null-null-GPN_EN-cpc-pf-google-wwe%2526utm_content%253DPCF8574%2526ds_k%253DPCF8574%2526DCM%253Dyes%2526gclid%253DEAIaIQobChMIt4yjzsG07wIVD-J3Ch2gYgz8EAAYASAAEgImWfD_BwE%2526gclsrc%253Daw.ds). 
But when reading the pins using `digitalRead(uint8_t pin);` it returns Low.

The read returns the correct state, only after at least one pin has been set.

xreef, 2021-03-16:
Hi jhengstermann,
to set the initial value to HIGH you must set pin to INPUT_PULLUP.
Bye Renzo

jhengstermann, 2021-03-17:
I think there is still a problem, or i'am misunderstanding something
I've added a couple of debug lines to the `digitalReadAll` function and this is the output:
```
10:39:37.003 -> W: 0 R ALL: 1 R Down: 0R Up: 1
10:39:37.003 -> Set pin 1 as 2
10:39:37.003 -> W: 0 R ALL: 11 R Down: 0R Up: 11
10:39:37.003 -> Set pin 2 as 2
10:39:37.003 -> W: 0 R ALL: 111 R Down: 0R Up: 111
10:39:37.003 -> Set pin 3 as 2
10:39:37.003 -> W: 0 R ALL: 1111 R Down: 0R Up: 1111
10:39:37.003 -> Set pin 4 as 2
10:39:37.003 -> W: 0 R ALL: 11111 R Down: 0R Up: 11111
10:39:37.003 -> Set pin 5 as 2
10:39:37.003 -> W: 0 R ALL: 111111 R Down: 0R Up: 111111
10:39:37.003 -> Set pin 6 as 2
10:39:37.003 -> W: 0 R ALL: 1111111 R Down: 0R Up: 1111111
10:39:37.003 -> Set pin 7 as 2
10:39:37.003 -> W: 0 R ALL: 11111111 R Down: 0R Up: 11111111
10:39:37.003 -> Read from buffer
10:39:37.003 -> Data ready
10:39:37.003 -> Input 11111111Buffer value 0
10:39:37.003 -> Buffer hight value readed set readed 0
10:39:37.003 -> Return value: 00000000
```
you can see that the byte read from i2c is B11111111, which is correct, because the pins are all high on startup.
But the return value is all 00000000

xreef, 2021-03-17:
Hi,
can you post your code.
Thanks Renzo

jhengstermann, 2021-03-18:
here it is:
```c
#define GPIO_EXP_ADR 0x38

#include <Arduino.h>
#include "PCF8574.h"

PCF8574 ioX_pcf8574(GPIO_EXP_ADR);

void setup() {
	Serial.begin(115200);
	Serial.println();
	delay(10);

  	ioX_pcf8574.begin();
	delay(10);

	for (int i=0; i<8; i++) {
		ioX_pcf8574.pinMode(i, INPUT_PULLUP);
	}
	delay(10);

	ioX_pcf8574.digitalReadAll();
}

void loop() {
}
```
and in `PCF8574.cpp` i just added these print statements to the end of `PCF8574::DigitalInput PCF8574::digitalReadAll(void)`
```c
DEBUG_PRINT(digitalInput.p0);
DEBUG_PRINT(digitalInput.p1);
DEBUG_PRINT(digitalInput.p2);
DEBUG_PRINT(digitalInput.p3);
DEBUG_PRINT(digitalInput.p4);
DEBUG_PRINT(digitalInput.p5);
DEBUG_PRINT(digitalInput.p6);
DEBUG_PRINTLN(digitalInput.p7);
```

im using an ESP8266.

xreef, 2021-03-18:
Hi jhengstermann,
I spent hours trying to debug your problem, couldn't figure out why it happened.
But in the end I didn't notice the most obvious thing.

You put the begin of the pcf8574 before the pins initialization: P.

Bye Renzo

jhengstermann, 2021-03-18:
yeah thanks 🙏

closed 2021-03-18 by jhengstermann