xreef/PCF8574_library #42: Drive a mosfet or BJT from PCF8574

tsctrl, 2021-03-21:
Hi @xreef,

Thank you for this great library and all the effort you did.

I am having difficulties to drive a mosfet or bjt using the PCF8574 GPIO as a basic switch. My objective is to turn on and off audio lines using the mosfet or bjt driven by the expender.

I have basic understanding on the source and sink current. have complete working circuit running on opto coupler driven by the expender without any issue. but i not able to do a direct basic bjt or mos switching using the expender. led have anode and cathode but transistor only have one base. so i am missing on that part where to sink the current on the single base pin.

do you have experience on this regards? simple connection guide will do help me alot.
thank!

xreef, 2021-03-24:
Hi tsctrl,
I use the pcf8574 with a 2n2222 and work correctly.
This is an NPN transistor and need a very low current, what kind of mosfet do you use?
Bye Renzo

tsctrl, 2021-04-01:
hi @xreef,

i am using s8050. unable to make it to work. do you have tutorial or can share the connection you did the with the 2n2222.

thank you.