xreef/PCF8574_library #44: Example interruptWemos crash

Tommamgo, 2021-05-03:
Hello, 
Every time I try the example interruptWemos on my Wemos D1 Mini, it crashes. At the beginning I thought I connected something wrong, but with the example LedWemos everything works. I suspect that for some reason something is wrong with the interrupt pin, but what? 

This is my error message:

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Abort called

>>>stack>>>

ctx: cont
sp: 3ffffef0 end: 3fffffc0 offset: 0000
3ffffef0:  feefeffe feefeffe feefeffe 00000100  
3fffff00:  000000fe 00000000 00000000 00000000  
3fffff10:  00000000 00000000 00000000 00ff0000  
3fffff20:  5ffffe00 5ffffe00 3ffef31c 00000000  
3fffff30:  00000002 0000000d 3ffee514 4020251a  
3fffff40:  40100686 2bde5da7 ffffff00 4020252c  
3fffff50:  40101a49 0004ea65 3ffee514 40203185  
3fffff60:  00000000 3ffee65c 000003e8 3ffee554  
3fffff70:  3fffdad0 3ffee65c 000003e8 3ffee554  
3fffff80:  3fffdad0 3ffee4ec 3ffee514 40203234  
3fffff90:  3fffdad0 3ffee4ec 3ffee514 40201075  
3fffffa0:  3fffdad0 00000000 3ffee514 40202128  
3fffffb0:  feefeffe feefeffe 3ffe84e8 40100f85  
<<<stack<<<

xreef, 2021-05-03:
Hi Tommamgo,
now the sketch works correctly? 
Bye Renzo

Tommamgo, 2021-05-03:
I'm not quite sure about that.... 
When I upload the sketch, I get this output.

...
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
Init pcf8574...OK
...

**First when I press a button it changes.** 

KEY0 PRESSED
KEY2 PRESSED
KEY3 PRESSED
KEY4 PRESSED
KEY5 PRESSED
KEY6 PRESSED
KEY7 PRESSED

sergiocntr, 2021-05-28:
in my opinion with esp family is always better put some delay in every loop or the watchdog will reset.

closed 2021-05-30 by xreef