xreef/PCF8574_library #45: Interrupt Mode

joko2020, 2021-06-05:
Hi There ,

Just want to ask , is that any ability Interrupt mode in this library? like HIGH, LOW, CHANGE, FALLING, RAISING.
as description in Arduino IDE we can use the attachInterrupt() function, that accepts as arguments: the GPIO pin, the name of the function to be executed, and mode:  `attachInterrupt(digitalPinToInterrupt(GPIO), function, mode);` 

Thanks

xreef, 2021-06-05:
Hi joko2020,
pcf8574 have an internal interrupt that monitor pin change, you can get more information on this article https://www.mischianti.org/2019/01/02/pcf8574-i2c-digital-i-o-expander-fast-easy-usage/

you can get an example here with core library interrupt management https://github.com/xreef/PCF8574_library/blob/master/examples/keyPressed_withInterrupt/keyPressed_withInterrupt.ino

If you need more info write here.

Bye RM

joko2020, 2021-06-06:
Ok thanks

closed 2021-06-06 by joko2020