xreef/PCF8574_library #46: latency and delay

tsctrl, 2021-06-13:
Hi @xreef ,
thanks for the great library,

how the library getLatency() function work? is it calculating twowire latency to transmit in realtime or is just retrieving latency setting in twowire? normally should put delay based on the latency setting to retrieve output?

thanks!

xreef, 2021-06-14:
Hi tsctrl,
the latency parameter work like a debounce, and It's managed by the library not from IC.
For example  if you se a latency of 1000 millis, and pin is set as INPUT_PULLUP the library read from IC only if the pin have state HIGH and if already LOW no more read from IC.
Prevent not usefully call.

If you need a realtime reading every time you must set latency to 0 or uncomment the define
// #define PCF8574_LOW_LATENCY

Bye Renzo

tsctrl, 2021-06-18:
Hi @xreef, 

1. ok, debounce means delay between between read, the realtime reading is only for input? how about output? do this setting effect the isLastTransmissionSuccess() result as the output are not realtime? is that possible.. 

2. one more, is it possible to redefine wire and pass in the class initializer as the same predefine wire in the class? i not sure if the wire already defined during boot. if that is possible what is the speed that can be use? and if i need to constantly read or write value such in a loop, what is max call in milis i could use with delays that work without calling the wire too much that can cause bus busy and invalid state?

thanks in advanced

xreef, 2021-06-19:
Hi,
No work only for read, and it used for reduce read and preserve state change until read.
Yes, it's what you need.
The limit is 100khz from specs of pcf8574.
Bye Renzo

tsctrl, 2021-06-19:
i am not even can run at rate of 50ms delay.  10K pullup 1000Hz rtos, keep getting error as below:

```
[502799][W][esp32-hal-i2c.c:1434] i2cCheckLineState(): invalid state sda(21)=0, scl(22)=1
[542128][E][esp32-hal-i2c.c:333] i2cDumpI2c(): i2c=0x3ffb1000
[542128][I][esp32-hal-i2c.c:334] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[542165][I][esp32-hal-i2c.c:336] i2cDumpI2c(): lock=0x3ffc842c
[542242][I][esp32-hal-i2c.c:338] i2cDumpI2c(): num=0
[542311][I][esp32-hal-i2c.c:339] i2cDumpI2c(): mode=1
[542379][I][esp32-hal-i2c.c:340] i2cDumpI2c(): stage=3
[542449][I][esp32-hal-i2c.c:341] i2cDumpI2c(): error=5
[542519][I][esp32-hal-i2c.c:342] i2cDumpI2c(): event=0x3ffaff28 bits=112
[542607][I][esp32-hal-i2c.c:343] i2cDumpI2c(): intr_handle=0x3ffaff5c
[542692][I][esp32-hal-i2c.c:344] i2cDumpI2c(): dq=0x3ffe6644
[542768][I][esp32-hal-i2c.c:345] i2cDumpI2c(): queueCount=1
[542843][I][esp32-hal-i2c.c:346] i2cDumpI2c(): queuePos=0
[542916][I][esp32-hal-i2c.c:347] i2cDumpI2c(): errorByteCnt=0
[542993][I][esp32-hal-i2c.c:348] i2cDumpI2c(): errorQueue=0
[543068][I][esp32-hal-i2c.c:349] i2cDumpI2c(): debugFlags=0x00000000
[543153][I][esp32-hal-i2c.c:326] i2cDumpDqData(): Debug Buffer not Enabled
[543243][I][esp32-hal-i2c.c:369] i2cDumpInts(): Debug Buffer not Enabled
[543332][I][esp32-hal-i2c.c:1153] i2cProcQueue(): Bus busy, reinit
```
look like i cant run the digitalWrite multiple time if current havent end. 
i am using something like this to wait for current write before executing another one. but no success.

```
while(!PCF8574->digitalWrite(BUZZER_PIN, LOW)){
          delay(50);
}
```
thanks

xreef, 2021-06-19:
Hi tsctrl,
now I'm not at home, but please open a topic on my forum 
https://www.mischianti.org/forums/forum/mischiantis-libraries/pcf8574-i2c-digital-i-o-expander/
and add the code you use for test, and It's also important the microcontrollor type.
When return (at the end of next week) i'm going to try.
Bye Renzo

tsctrl, 2021-06-20:
sure, thanks @xreef

closed 2021-06-20 by tsctrl