xreef/PCF8574_library #47: How to Prevent Bouncing / Debounce mode

joko2020, 2021-06-15:
Helo @xreef ,

Is that any Function or config to Prevent bounce?,
I've try your example , but I have problem with bounce 
I used ESP32 Dev Module  with Arduino IDE 1.8.12.

Could you give and example sketch ?
I Use Pin 0 - 7 as input and also use INT pin to be external ISR for ESP32.

I also implemented your Interrupt Wemos Example but went the button is pressed It is not exact same condition time to time..

Thanks

xreef, 2021-06-15:
Hi yoko,
you can uncomment this 
// #define PCF8574_LOW_LATENCY
Or set latency to 0
or read with true as parameter

int p1Digital = PCF8574.digitalRead(P1, true); // read P1 without debounce

You can find more information here
https://www.mischianti.org/2019/01/02/pcf8574-i2c-digital-i-o-expander-fast-easy-usage/

Bye Renzo

joko2020, 2021-06-15:
Thanks for fast respond ,  firs I'll Try to uncomment  the LOW LATENCY..

and I will report again

joko2020, 2021-06-15:
@xreef  Two method that you had explain did not work as expected ,

I'm not sure, maybe stranger behavior came from my waveshare pcf8574 module 
because when I scan with I2C Scanner the Address is found  in 0x20.
but if I adjust the address in your **Keypressed_Withinterrupt** Example sketch to 0x20 my push button has bouncing 
but if use the default value of 0x39 in those sketch it's work as expected 

this is modified sketch that work as I need 
`
/*
 KeyPressed with interrupt
 by Mischianti Renzo <http://www.mischianti.org>

 https://www.mischianti.org/2019/01/02/pcf8574-i2c-digital-i-o-expander-fast-easy-usage/
*/

#include "Arduino.h"
#include "PCF8574.h"

// For arduino uno only pin 1 and 2 are interrupted
#define ARDUINO_UNO_INTERRUPTED_PIN 12
#define led 2
// Function interrupt
void keyPressedOnPCF8574();

// Set i2c address
PCF8574 pcf8574(0x39, ARDUINO_UNO_INTERRUPTED_PIN, keyPressedOnPCF8574);
unsigned long timeElapsed;
void setup()
{
	Serial.begin(115200);
	delay(1000);
  
	pinMode(led, OUTPUT);
	pcf8574.pinMode(P1, INPUT);
	Serial.print("Init pcf8574...");
	if (pcf8574.begin()){
		Serial.println("OK");
	}else{
		Serial.println("KO");
	}

  digitalWrite(led, HIGH);
	timeElapsed = millis();
}

bool keyPressed = false;
void loop()
{
	if (keyPressed){
		int val = pcf8574.digitalRead(P1,true);
		Serial.print("READ VALUE FROM PCF ");
		Serial.println(val);
    digitalWrite(led, !digitalRead(led));
    delay(120);
    keyPressed= false;
	}
}
void ledstate(){
  digitalWrite(led, !digitalRead(led));
}
void keyPressedOnPCF8574(){
	// Interrupt called (No Serial no read no wire in this function, and DEBUG disabled on PCF library)
	 keyPressed = true;

}
  
`

xreef, 2021-06-15:
It's very strange behavior, I think I can't re-generate that situation.
If you find more information write here and we try to do somethimg more.
Bye Renzo

joko2020, 2021-06-15:
I Found kind of BUG ,
If  I use statement like `PCF8574 pcf8574(0x39, ARDUINO_UNO_INTERRUPTED_PIN, keyPressedOnPCF8574); ` what ever i use the I2C address it still work.. I mean my ESP32 still can communicate with PCF8574 .

But if Write the statement like `PCF8574 pcf8574(0x39);
`   communication fail and I must change and use 0x20 as default address of PCF8574 for the program to run.

I found this situation in your example Keypressed_WithInterrupt  and InterruptWemos.

Maybe if you don't mind you can try  it. 

Thanks

xreef, 2021-06-16:
Hi joko,
damn It's very strange, I'm going to try, but I need some time.
Thanks again for the alert

joko2020, 2021-06-16:
ok thanks for your attention

xreef, 2021-11-24:
Hi Joko,
sorry for the late response, but I do all the test described and for me seems work all without problem.
Sorry again bye Renzo

closed 2021-11-24 by xreef