xreef/PCF8574_library #48: problem with ssd1306

alirezaimi, 2021-06-23:
Hi
Is there any known problem with ssd1306 ? 
after `Pinmode` and `begin` for this module, the ssd1306 commands not work at all !

I use `Adafruit_SSD1306` library for my 0.96 oled and your library for my pcf8574 .

xreef, 2021-06-24:
Hi Alizeraimi,
you use pcf with interrupt??
Bye Renzo

alirezaimi, 2021-06-24:
No @xreef, for now just using other ports.
I searched little bit and find out something related to clock speed, is this right ? If it's true how can I change these two to work together ?

xreef, 2021-06-24:
Hi Alirezaimi,
I use ssd1306 with pcf8574 anche I had problem only when use interrupt, in the nexts weeks I'm going to write an article on ssd1306 and pcf8574 with interrupt and I check better if there are other problems.

I use this code to work all together.
```cpp
	if (changed){
		SERIAL_DEBUG.print("ENCODER --> ");
		SERIAL_DEBUG.print(encoderValue);
		SERIAL_DEBUG.print(" - BUTTON --> ");
		SERIAL_DEBUG.println(encoderButtonVal?"HIGH":"LOW");

		pcf8574.detachInterrupt();
		if (changedButton) {
			currentScreen = clickMenu(currentScreen, encoderValue);
			changedButton = false;
		}

		renderScreen(currentScreen);

		display.clearDisplay();

		pcf8574.attachInterrupt();
		changed = false;
		pcf8574.readEncoderValue(ENCODER_PIN_A, ENCODER_PIN_B);
		pcf8574.digitalRead(ENCODER_BUTTON);
	}
```

i use Adafruit library too, when I have more information I write here.

Bye Renzo

alirezaimi, 2021-06-24:
Thanks, but what can I do for solving my problem ????? Changing clock speed is the solution or not ?

xreef, 2021-06-25:
Hi alirezaimi,
specify the platform you use and put your code here.
https://www.mischianti.org/forums/forum/yours-project/

I'm going to check your problem. I try locally, and for me work without problem, but I try to do more test.

Bye Renzo

alirezaimi, 2021-06-25:
After I changed my code and test some pints on it, i find out that because i using another ports than default port that declared in adafriut library for ssd1306 and wire library (d4, d5), and i using d6 and d7 ports instead , my project not worked,

If i put `Wire.begin()` before pcf8574 pinmodes the ssd1306 not work, but if i put it after pinmodes its worked! 

This code not work :

    void setup() {
      Serial.begin(115200);
      delay(10);

    // oled settings
      Wire.begin(OLED_SDA, OLED_SCK);  // oled change default i2c pins from d4 and d5  

      expander.pinMode(P0, INPUT_PULLUP); // touch 1
      expander.pinMode(P1, OUTPUT);  // relay 1
      expander.pinMode(P2, OUTPUT); // relay 2
      expander.pinMode(P3, INPUT_PULLUP); // touch 2
      expander.begin();

      
      // SSD1306_SWITCHCAPVCC = generate display voltage from 3.3V internally
      if(!display.begin(SSD1306_SWITCHCAPVCC, OLED_SCREEN_ADDRESS)) {
        Serial.println(F("SSD1306 allocation failed"));
        for(;;); // Don't proceed, loop forever
      }
      // Show initial display buffer contents on the screen --
      // the library initializes this with an Adafruit splash screen.
      display.display();
      // Clear the buffer
      display.clearDisplay();
      // Draw a single pixel in white
      display.drawPixel(10, 10, SSD1306_WHITE);
      // Show the display buffer on the screen. You MUST call display() after
      // drawing commands to make them visible on screen!
      display.display();
      ...

and this one worked :

    void setup() {
      Serial.begin(115200);
      delay(10);

      expander.pinMode(P0, INPUT_PULLUP); // touch 1
      expander.pinMode(P1, OUTPUT);  // relay 1
      expander.pinMode(P2, OUTPUT); // relay 2
      expander.pinMode(P3, INPUT_PULLUP); // touch 2
      expander.begin();

    // oled settings
      Wire.begin(OLED_SDA, OLED_SCK);  // oled change default i2c pins from d4 and d5  
      
      // SSD1306_SWITCHCAPVCC = generate display voltage from 3.3V internally
      if(!display.begin(SSD1306_SWITCHCAPVCC, OLED_SCREEN_ADDRESS)) {
        Serial.println(F("SSD1306 allocation failed"));
        for(;;); // Don't proceed, loop forever
      }
      // Show initial display buffer contents on the screen --
      // the library initializes this with an Adafruit splash screen.
      display.display();
      // Clear the buffer
      display.clearDisplay();
      // Draw a single pixel in white
      display.drawPixel(10, 10, SSD1306_WHITE);
      // Show the display buffer on the screen. You MUST call display() after
      // drawing commands to make them visible on screen!
      display.display();
      ...

alirezaimi, 2021-06-25:
And now pcf not work anymore :(

xreef, 2021-06-25:
Hi Alirezaimi,
check the documentation on https://www.mischianti.org/2019/01/02/pcf8574-i2c-digital-i-o-expander-fast-easy-usage/
and use correct pcf constructor.
If you use different sda scl you must specify different SDA SCL on pcf.

`PCF8574(uint8_t address, uint8_t sda, uint8_t scl);`

Bye Renzo

closed 2021-06-25 by xreef

alirezaimi, 2021-07-02:
My bad :((((( 
I change a port of i2c by mistake .
thanks a lot .