xreef/PCF8574_library #55: Error when compiling.

limaomerces, 2022-04-24:
Hello. I'm using Esp32 and I used its library with the TwoWire example.

- I didn't edit
- I didn't make any changes
- I didn't add anything.
- Just opened the example and loaded it.

Got this error when compiling:

```
C:\Users\Tiago\Documents\Arduino\libraries\PCF8574_library-master\examples\ledEsp32OnTheSecondI2C\ledEsp32OnTheSecondI2C.ino: In function 'void setup()':
ledEsp32OnTheSecondI2C:47:28: error: call of overloaded 'begin(int, int, int)' is ambiguous
   I2Cone.begin(16,17,400000); // SDA pin 16, SCL pin 17, 400kHz frequency
                            ^
In file included from C:\Users\Tiago\Documents\Arduino\libraries\PCF8574_library-master/PCF8574.h:38,
                 from C:\Users\Tiago\Documents\Arduino\libraries\PCF8574_library-master\examples\ledEsp32OnTheSecondI2C\ledEsp32OnTheSecondI2C.ino:30:
C:\Users\Tiago\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.2\libraries\Wire\src/Wire.h:79:10: note: candidate: 'bool TwoWire::begin(int, int, uint32_t)'
     bool begin(int sda=-1, int scl=-1, uint32_t frequency=0); // returns true, if successful init of i2c bus
          ^~~~~
C:\Users\Tiago\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.2\libraries\Wire\src/Wire.h:80:10: note: candidate: 'bool TwoWire::begin(uint8_t, int, int, uint32_t)'
     bool begin(uint8_t slaveAddr, int sda=-1, int scl=-1, uint32_t frequency=0);
          ^~~~~
exit status 1
call of overloaded 'begin(int, int, int)' is ambiguous
`
```

avillacis, 2022-04-25:
This is due to the poorly-implemented overload that implements I2C slave support in 2.0.2:

https://github.com/espressif/arduino-esp32/issues/6616

In this particular case, the example sketch *might* be compile-fixed by using 400000U instead of 400000, in order to force the compiler to choose the overload with the explicitly-unsigned frequency parameter.

xreef, 2022-04-26:
Hi,
yes, thanks to @avillacis for the tips.
Bye Renzo

closed 2022-04-26 by xreef