xreef/PCF8574_library #56: Lockup on PCF8574::begin() on ESP32 due to unpaired beginTransmission() call

avillacis, 2022-04-25:
When compiling and running a sketch under arduino-esp32 v1.0.6, the library works OK. However, when using v2.0.2 or later, the sketch locks up completely inside the PCF8574::begin() call.

By digging into the code, I found out that there is an unpaired beginTransmission() call on the _wire ptr, that is, a beginTransmission not paired with a corresponding endTransmission() call on the same instance:

The two conflicting calls to beginTransmission() are:

https://github.com/xreef/PCF8574_library/blob/59cf47ccff8b22c5a57b17ba3649b851f18bbe9d/PCF8574.cpp#L200

https://github.com/xreef/PCF8574_library/blob/59cf47ccff8b22c5a57b17ba3649b851f18bbe9d/PCF8574.cpp#L211

The single call to endTransmission():
https://github.com/xreef/PCF8574_library/blob/59cf47ccff8b22c5a57b17ba3649b851f18bbe9d/PCF8574.cpp#L219

In context:
https://github.com/xreef/PCF8574_library/blob/59cf47ccff8b22c5a57b17ba3649b851f18bbe9d/PCF8574.cpp#L198-L220

If one of the two calls is removed (I chose the one at line 211), the sketch no longer locks up.

xreef, 2022-04-26:
Thanks, @avillacis,
I'm going to fix it in the next release.
Thanks again Renzo

xreef, 2022-04-27:
Thanks again

closed 2022-04-27 by xreef