xreef/PCF8574_library #57: Got digitalReadAll to work, but digitalRead is not?

we9v, 2022-05-01:
After a bit of fumbling around (hardware engineer, not software), I finally got digitalReadAll to work to read 4 button inputs, but immediately trying to do digitialRead on those 4 pins returns nothing.
Code is over 1600 lines long, so here is some snippets of the pcf code:
```
#define pcfButtonLeft   5          //SYSTEM PARAMETER - PCF8574 pin for button left
#define pcfButtonRight  4          //SYSTEM PARAMETER - PCF8574 pin for button right
#define pcfButtonBack   6          //SYSTEM PARAMETER - PCF8574 pin for button back
#define pcfButtonSelect 7          //SYSTEM PARAMETER - PCF8574 pin for button select
#define pcfBuzzer       0          //SYSTEM PARAMETER - PCF8574 pin for buzzer, active low
#define pcfLedRed       1          //SYSTEM PARAMETER - PCF8574 pin for Red LED, active low
#define pcfLedGreen     2          //SYSTEM PARAMETER - PCF8574 pin for Green LED, active low
#define pcfLedBlue      3          //SYSTEM PARAMETER - PCF8574 pin for Blue LED, active low
#define PCF8574_INTERRUPT_PIN    19          //SYSTEM PARAMETER - Button Interrupt Pin (PCF8574)
#include "PCF8574.h"                // https://github.com/xreef/PCF8574_library for I2C expander
void keyPressedOnPCF8574();          // for use with interrupt
PCF8574 pcf8574(0x20, PCF8574_INTERRUPT_PIN, keyPressedOnPCF8574);  // initialize with I2C address
```
In setup

```
//PCF8574 INITALIZATION
  pcf8574.pinMode(pcfBuzzer, OUTPUT, HIGH);       // Buzzer
  pcf8574.pinMode(pcfLedRed, OUTPUT, HIGH);       // Red of RGB LED
  pcf8574.pinMode(pcfLedGreen, OUTPUT, HIGH);     // Green of RGB LED
  pcf8574.pinMode(pcfLedBlue, OUTPUT, HIGH);      // Blue of RGB LED

  pcf8574.pinMode(pcfButtonSelect, INPUT);        // SELECT button
  pcf8574.pinMode(pcfButtonBack, INPUT);          // BACK button
  pcf8574.pinMode(pcfButtonLeft, INPUT);          // LEFT button
  pcf8574.pinMode(pcfButtonRight, INPUT);         // RIGHT button

  Serial.print("Init pcf8574...");
  if (pcf8574.begin()){
    Serial.println("OK");
  }else{
    Serial.println("NOT OK");
  }

```

The code in question:
```
      PCF8574::DigitalInput pcfData = pcf8574.digitalReadAll();
      Serial.print(" PCFa:");   Serial.print(pcfData.p7);Serial.print(pcfData.p6);Serial.print(pcfData.p5);Serial.print(pcfData.p4);
      uint8_t val7 = pcf8574.digitalRead(P7);
      uint8_t val6 = pcf8574.digitalRead(P6);
      uint8_t val5 = pcf8574.digitalRead(P5);
      uint8_t val4 = pcf8574.digitalRead(P4);      
      Serial.print(" PCFr:");  Serial.print(val7);   Serial.print(val6);  Serial.print(val5);  Serial.print(val4);
```

Buttons have pullups, and the PCF pins will read zero/ground when buttons are pressed.  This is running in a loop, and here are the results, initially with no buttons pressed, then pressing all of them, and then releasing all of them.  You can see that the digitalReadAll ("PCFa") values do go low when I press the buttons, but the digitalRead ("PCFr") values are stuck at 0's.
```
 PCFa:1111 PCFr:0000
 PCFa:1111 PCFr:0000
 PCFa:0111 PCFr:0000
 PCFa:0111 PCFr:0000
 PCFa:0101 PCFr:0000
 PCFa:0000 PCFr:0000
 PCFa:0000 PCFr:0000
 PCFa:0000 PCFr:0000
 PCFa:0000 PCFr:0000
 PCFa:0000 PCFr:0000
 PCFa:0000 PCFr:0000
 PCFa:0000 PCFr:0000
 PCFa:0000 PCFr:0000
 PCFa:1111 PCFr:0000
 PCFa:1111 PCFr:0000
 PCFa:1111 PCFr:0000
 PCFa:1111 PCFr:0000
```

we9v, 2022-05-01:
I have discovered if I comment out the two lines with the digitalReadAll:
 ```
//PCF8574::DigitalInput pcfData = pcf8574.digitalReadAll();
//Serial.print(" PCFa:");   Serial.print(pcfData.p7);Serial.print(pcfData.p6);Serial.print(pcfData.p5);Serial.print(pcfData.p4);

```
Then the individual digitalRead's work.   Does the use of digitialReadAll preclude/void the use of digitialRead?

I've also now found out that if I do the digitalReadAll AFTER the individual digitalReads, then it also continues to work.
So why does the original code/order not work?   Is there some minimum time between a digitalReadAll and digitalRead?  My loop operates at about 260ms.  So in one order of events, digitalRead comes immediately after ReadAll, and the other order has about 260ms after the ReadAll before the digitalRead.

xreef, 2022-05-01:
Hi @we9v,
I use a denounce to manage a multiple key press.. 
To bypass denounce you can pass true to the read.
You can also reduce to 0 the denounce time.

The difference with figitalReadAll is that this method doesn't has denounce.
Bye Renzo

we9v, 2022-05-01:
So, I can get it to work by passing true, but I don't understand why it returns the wrong value (always reads 0) if I don't pass true.  How am I to know if digitalRead is passing the correct value? (besides always passing true)

xreef, 2022-05-02:
If you use only digitalRead or digitalReadAll I think the problem disappear because digitalRead do the retention of the value, but digitalReadAll reset that retention to the initial value.
Bye Renzo

closed 2022-05-10 by xreef