xreef/PCF8574_library #58: Do library examples work with ESP32 2.02 ??

MnemonicFish, 2022-07-04:
Tried ESP32-DevKit3, DOIT DevkitV1 ?? confirmed 0x20 I2C address, using default GPIO pins SDA, 21 SCL, 22 

receiving "KO" messaged on pcf8574.begin();  which I have read is common, 
 
Thanks in advance!

//----------------------------------------------
#include "Arduino.h"
#include "PCF8574.h"

// Set i2c address
PCF8574 pcf8574(0x20);

unsigned long timeElapsed;
void setup()
{
  Serial.begin(115200);
  delay(1000);

  pcf8574.pinMode(P0, OUTPUT);
  pcf8574.pinMode(P1, INPUT);

  Serial.print("Init pcf8574...");
  if (pcf8574.begin()){
    Serial.println("OK");
  }else{
    Serial.println("KO");
  }

}

void loop()
{
  uint8_t val = pcf8574.digitalRead(P1);
  if (val==HIGH) Serial.println("KEY PRESSED");
  delay(50);
}

xreef, 2022-07-06:
Hi @MnemonicFish,
I don't know, try to check the wiring.

MnemonicFish, 2022-07-06:
It's puzzling because I verified correct wiring (see image below) by running the WIRE library sketch below and it works fine, flashing all LEDs on/off. 

When I upload the PCF8574_library example (see below) code after the Wire.h sketch, nothing changes, the shift register stays the same (all LEDS remain in the flashing on/off cycle). 

//------------START OF -------- WORKING CODE using Wire.h  ----------------------

#include<Wire.h>

//#define P0         0x01     //B00000001
//#define P1         0x02     //B00000010
//#define P2         0x04     //B00000100
//#define P3         0x08     //B00001000
//#define P4         0x10     //B00010000
//#define P5         0x20     //B00100000
//#define P6         0x40     //B01000000
//#define P7         0x80     //B10000000
//#define All_off   0x00     //B00000000  
//#define All_on    0xff      //B11111111 

void setup()
{
  Serial.begin(115200);
  Wire.begin();

  Wire.beginTransmission(0x20);
  Wire.write(0x00);                 // turn OFF all LEDs
  Wire.endTransmission();
}

void loop()
{
  Wire.beginTransmission(0x20);
  Wire.write(0xff);                 // turn ON all LEDs
  Wire.endTransmission();
  
  delay(1000);

  Wire.beginTransmission(0x20);
  Wire.write(0x00);                 // turn OFF all LEDs
  Wire.endTransmission();
  
  delay(1000);
}  

//------------END OF --- WORKING CODE using Wire.h  ------------------------

[image]

Can you kindly confirm? 
1. are there addition steps required for configuration,  i.e, uncomment any lines of code in .cpp or h files?  
2. is specific initializing required to use ESP32 GPIO 21 and 22 for SDA/ SCL?  

Thanks!
f

xreef, 2022-07-08:
Hi,
I think It's all ok, It's very strange.
If I find some time I try to check.
Bye Renzo

yak1ex, 2022-07-09:
I'm a quite newbie, so there may be misunderstandings.
Anyway, similar behavior is observed here also, and initializing `Wire` before `begin()` eliminates the error.

```arduino
PCF8574 pcf8574(0x20);
void setup(void) {
    Serial.begin(115200);
    Wire.begin(21,22); // Without this line, KO is returned
    delay(1000);
    Serial.println("INIT");

    pcf8574.pinMode(P4, OUTPUT);
    pcf8574.pinMode(P5, INPUT);
    Serial.print("Init pcf8574...");
    if(pcf8574.begin()) {
        Serial.println("OK");
    } else {
        Serial.println("KO");
    }
}
```

MnemonicFish, 2022-07-09:
I tried inserting Wire.begin() and using the pcf8574(_address, _sda, _scl) constructor without success.

like you say the use of wire.begin() removes the KO error but the shifter is not updating the LEDs,, debugging shows that programmatically data is being sent but no physical change of LEDs, which have been tested every time and found functioning with the same shifter and wiring using the Wire sketch above,  

I can only assume that there is a software configuration issue and the data is not being transferred using GPIO 21 (SDA) and GPIO 22 (SCL). 

//---------------- CODE -----------------------------------------//

#include "Arduino.h"
#include "PCF8574.h"

//SDA: SDA (default is GPIO 21)
//SCL: SCL (default is GPIO 22)

PCF8574 pcf8574(0x20, 21, 22);

void setup()
{
  Serial.begin(115200);
  Wire.begin();
  delay(1000);

  pcf8574.pinMode(P0, OUTPUT);
  pcf8574.pinMode(P1, OUTPUT);
  pcf8574.pinMode(P2, OUTPUT);
  pcf8574.pinMode(P3, OUTPUT);
  pcf8574.pinMode(P4, OUTPUT);
  pcf8574.pinMode(P5, OUTPUT);
  pcf8574.pinMode(P6, OUTPUT);
  pcf8574.pinMode(P7, OUTPUT);

  Serial.print("Init pcf8574...");
  if (pcf8574.begin()) {
    Serial.println("OK");
  } else {
    Serial.println("KO");
  }
}

void loop()
{
  pcf8574.digitalWrite(P0, HIGH);
  delay(50);
  pcf8574.digitalWrite(P0, LOW);
}
//------------- END CODE -------------------------------//

//----------- SERIAL MONITOR  DEBUGGING--------------------------------//

Begin trasmission
Value 0 Write data pre 11111111Write data 11111110 for pin 0 bin value 1 value 0 byteBuffered 11111110
Going to write data 11111110
Start end trasmission if stop here check pullup resistor.
Begin trasmission
Value 1 Write data pre 11111110Write data 11111111 for pin 0 bin value 1 value 1 byteBuffered 11111111
Going to write data 11111111
Start end trasmission if stop here check pullup resistor.
Begin trasmission
Value 0 Write data pre 11111111Write data 11111110 for pin 0 bin value 1 value 0 byteBuffered 11111110
Going to write data 11111110
Start end trasmission if stop here check pullup resistor.
Begin trasmission
Value 1 Write data pre 11111110Write data 11111111 for pin 0 bin value 1 value 1 byteBuffered 11111111
Going to write data 11111111
Start end trasmission if stop here check pullup resistor.
Begin trasmission
Value 0 Write data pre 11111111Write data 11111110 for pin 0 bin value 1 value 0 byteBuffered 11111110
Going to write data 11111110
Start end trasmission if stop here check pullup resistor.

//------------------ END DEBUGGING ----------------------------------------------//

xreef, 2022-07-11:
Hi all,
It's interesting, I'm going to investigate soon.
Bye Renzo

MnemonicFish, 2022-07-12:
That's great Renzo, 

I'm going to run some more testing to see if using wire.begin()in this way might cause any conflicts... 

thank you for the library, appreciate your dedication!

xreef, 2022-07-28:
I finally fixed It, the problem is the type of sda and scl I use uint8_t but now esp32 take the address with uint8_t and use int for sda scl.
Thank @yak1ex @MnemonicFish @bvp @thijstriemstra  for the help and patience.
Bye Renzo

closed 2022-07-28 by xreef