xreef/PCF8574_library #60: Won't compile with ESP8266 DevKit 1

jsmwrench, 2022-12-31:
Hello.

I can't seem to get any of your examples to compile using the Arduino IDE.  I'm using and ESP8266 Node MCU dev kit 1 board and when I attempt to compile any of your given examples I get the following error:

`c:\Users\jsmwr\Documents\Arduino\libraries\PCF8574\PCF8574.cpp: In member function 'bool PCF8574::begin()':
c:\Users\jsmwr\Documents\Arduino\libraries\PCF8574\PCF8574.cpp:195:36: error: no matching function for call to 'TwoWire::begin(int, int)'
  195 |   _wire->begin((int)_sda, (int)_scl);`

Any assistance would be appreciated.  Thank you!
Ron.

xreef, 2023-01-02:
Hi,
sorry I lost this issue, I try to check it soon.
Can you add the link to your device and your Arduino configuration.
Bye Renzo

jsmwrench, 2023-01-02:
Hi!  
I appreciate that you responded.  However, I found out that my "board" libraries were corrupted within my IDE.  After a fresh install, it all worked.  Thank you and Happy New Year!
Ron.

closed 2023-01-02 by jsmwrench