xreef/PCF8574_library #61: Both digitalReadAll and digitalRead return same value for all INPUT or INPUT_PULLUP pins

cjkarande, 2023-03-24:
I am using a PCF8574 module along with ESP8266 to read touch TP223 base Touch sensors with PCF8574's Interrupt mode

The interrupt seems to work fine i.e. it fires whenever I touch the sensor connected to any of the P0 to P7 pins. However, the digitalRead returns 0 for all pins when configured as INPUT and it returns 1 if they are configured as INPUT_PULLUP. Same is the case with digitalReadAll. I am not able to detect a state change of the touch sensors. I have also tried reading all inputs in the loop function, with both Read and ReadAll but still no difference.

Setup details:
PCF8574 library version: 2.3.5
MCU: ESP8266-12F
SDA pin: D4
SCL pin: D2

//Initialization code. Tried with
for (uint8_t pin = 0; pin < 8; pin++)
      pcf8574.pinMode(pin, INPUT);
 pcf8574.begin();

//I tried this code
PCF8574::DigitalInput di = pcf8574.digitalReadAll();

// and also this code
for (pin = P0; pin < 8; pin++)
    if (pcf8574.digitalRead(pin)) {
        Serial.println(pin);
        break;
    }

Am I doing something wrong...

xreef, 2023-03-24:
Hi @cjkarande,
do another test, try 

`
// and also this code
for (byte pin = P0; pin <= P7; pin++)
Serial.print(pcf8574.digitalRead(pin));
`

Bye Renzo

cjkarande, 2023-03-24:
@xreef of course i had tried with the default digitalRead as well, but it same results...Interestingly, as mentioned, the interrupt seems to work fine. Issue is with digitalRead (both true & false) and digitalReadAll returning same value before and after touch. 

Am a bit lost on this issue, hope you can show me some way....
Thanks

cjkarande, 2023-03-26:
Hello @xreef, any lead on this ...what do you think may be going wrong? Would it be a hardware issue...

cjkarande, 2023-04-01:
Hello @xreef, appreciate any inputs from your side...This is a show stopper issue for me.

cjkarande, 2023-04-05:
Anybody else in the community who can help on this issue....

xreef, 2023-04-14:
Hi,
I try several test but all works.
I can't replicate your problem, I think the problem was hardware.
Bye Renzo

closed 2023-04-16 by xreef