xreef/PCF8574_library #62: aduino web editor

teberik, 2023-04-16:
I have been using the PCF8574_library on my Windows computer with the Arduino Editor application, and it has been working without any issues. However, when I tried to use the same library on the Arduino Web Editor, I encountered an error message even though the PCF8574 library is already added.

I was wondering if you could help me resolve this issue. Is there anything specific that I need to do in order to make the PCF8574_library work on the web editor? I would greatly appreciate any guidance or suggestions you may have.

Thank you very much for your help in advance.

xreef, 2023-04-16:
Hi teberik,
It's possible that the IDE import automatically another library, try to include like so
`#include "PCF8575_library.h"`
Please, give me feedback.
Thanks, Renzo

teberik, 2023-04-16:
> 

Yes, it works this way. Thank you very much.

closed 2023-04-16 by teberik