xreef/PCF8574_library #65: OUTPUT SET TO HIGH DURING BEGIN

edg2411, 2023-08-17:
Hello Im having an issue as it says in the title is there any reason why even though I set the output to low before calling begin() during the begin function it drives the input to high and I need to set it back to low after begin ends. The only thing that may be different from an example from your library is that Im also using an RTC but I dont think that have something to do with my issue.
Thanks in advance and great library btw!

xreef, 2023-08-18:
Hi edg,
How do you declare the pin INPUT??
Bye Renzo

edg2411, 2023-08-18:
It is an output,
`  for(int i=0;i<8;i++) {
    pcf8574.pinMode(i, OUTPUT);
  }
  pcf8574.pinMode(P2, OUTPUT);
	pcf8574.pinMode(P1, INPUT_PULLUP);
	pcf8574.pinMode(P0, INPUT_PULLUP);
  pcf8574.digitalWrite(P2, LOW)

  serialMonitor.sendStatusMessage("Init pcf8574...");
  if (pcf8574.begin()){
    serialMonitor.sendStatusMessage("OK\n");
  } else {
    serialMonitor.sendStatusMessage("KO\n");
  }
  pcf8574.digitalWrite(P2, LOW);`

it start low during begin at some point the output is set to high and if I dont drive it low again it stays high

xreef, 2023-08-26:
Hi,
read the documentation [here](https://mischianti.org/2019/01/02/pcf8574-i2c-digital-i-o-expander-fast-easy-usage/#InputOutput_mode_and_starting_value).
Bye Renzo

closed 2023-08-26 by xreef