xreef/PCF8574_library #66: Compilation error: expected initializer before 'updateEncoder'

galagen8, 2023-12-13:
Hello! I used this example from library:

```
#include "Arduino.h"
#include "PCF8574.h"

int encoderPinA = P0;
int encoderPinB = P1;

#define INTERRUPTED_PIN D7

void ICACHE_RAM_ATTR updateEncoder();

// initialize library
PCF8574 pcf8574(0x38, INTERRUPTED_PIN, updateEncoder);

volatile long encoderValue = 0;
uint8_t encoderButtonVal = HIGH;

void setup()
{
	Serial.begin (9600);
	delay(500);

	// encoder pins
	pcf8574.encoder(encoderPinA, encoderPinB);
	// encoder button
	pcf8574.pinMode(P2, INPUT);

	// Start library
	Serial.print("Init pcf8574...");
	if (pcf8574.begin()){
		Serial.println("OK");
	}else{
		Serial.println("KO");
	}

}

bool changed = false;

// The loop function is called in an endless loop
void loop()
{
	if (changed){
		Serial.print("ENCODER --> ");
		Serial.print(encoderValue);
		Serial.print(" - BUTTON --> ");
		Serial.println(encoderButtonVal?"HIGH":"LOW");
		changed = false;
	}
}

bool valPrecEncoderButton = LOW;
void updateEncoder(){
	changed = pcf8574.readEncoderValue(encoderPinA, encoderPinB, &encoderValue);

//	int vale = pcf8574.readEncoderValue(encoderPinA, encoderPinB);
//	if (vale!=0){
//		changed = true;
//	}
//	encoderValue = encoderValue + vale;

	// Button management
	encoderButtonVal = pcf8574.digitalRead(P2);
	if (encoderButtonVal!=valPrecEncoderButton){
	  changed = true; // Chnged the value of button
	  valPrecEncoderButton = encoderButtonVal;
	}
}

```

 but it showed me this error when compiling:

```
C:\Users\66842\AppData\Local\Temp\.arduinoIDE-unsaved20231113-10584-hmghi9.4faf9\encoderWithFullLibraryFunction\encoderWithFullLibraryFunction.ino:28:22: error: expected initializer before 'updateEncoder'
 void ICACHE_RAM_ATTR updateEncoder();
                      ^~~~~~~~~~~~~
C:\Users\66842\AppData\Local\Temp\.arduinoIDE-unsaved20231113-10584-hmghi9.4faf9\encoderWithFullLibraryFunction\encoderWithFullLibraryFunction.ino:26:25: error: 'D7' was not declared in this scope
 #define INTERRUPTED_PIN D7
                         ^
C:\Users\66842\AppData\Local\Temp\.arduinoIDE-unsaved20231113-10584-hmghi9.4faf9\encoderWithFullLibraryFunction\encoderWithFullLibraryFunction.ino:31:23: note: in expansion of macro 'INTERRUPTED_PIN'
 PCF8574 pcf8574(0x38, INTERRUPTED_PIN, updateEncoder);
                       ^~~~~~~~~~~~~~~
C:\Users\66842\AppData\Local\Temp\.arduinoIDE-unsaved20231113-10584-hmghi9.4faf9\encoderWithFullLibraryFunction\encoderWithFullLibraryFunction.ino:26:25: note: suggested alternative: 'A7'
 #define INTERRUPTED_PIN D7
                         ^
C:\Users\66842\AppData\Local\Temp\.arduinoIDE-unsaved20231113-10584-hmghi9.4faf9\encoderWithFullLibraryFunction\encoderWithFullLibraryFunction.ino:31:23: note: in expansion of macro 'INTERRUPTED_PIN'
 PCF8574 pcf8574(0x38, INTERRUPTED_PIN, updateEncoder);
                       ^~~~~~~~~~~~~~~
C:\Users\66842\AppData\Local\Temp\.arduinoIDE-unsaved20231113-10584-hmghi9.4faf9\encoderWithFullLibraryFunction\encoderWithFullLibraryFunction.ino:31:40: error: 'updateEncoder' was not declared in this scope
 PCF8574 pcf8574(0x38, INTERRUPTED_PIN, updateEncoder);
                                        ^~~~~~~~~~~~~

exit status 1

Compilation error: expected initializer before 'updateEncoder'

```

Could you help me where is a mistake and how to fix it? I'm already out of ideas

xreef, 2023-12-16:
Hi,
Your device is an esp8266?
Bye Renzo

closed 2023-12-17 by xreef

galagen8, 2023-12-18:
No, I use Arduino Nano CH340

xreef, 2023-12-18:
The sketch is for esp8266, use an example for Arduino.
Bye Renzo