xreef/PCF8574_library #71: Reconfig SDA & SCL to NON Default pins on ESP32-CAM AI-Thinker board, Why is this NOT Working for PCF8574 Library

MAWMN, 2024-12-20:
gives problems with the PCF8574 while the SSD display works fine with it: 
     Monitoring loop Started..
     Init pcf8574...OK
but after constantly changing the content of the display (Seconds wise), all interrupts of the PCF8574 gives :
     [ 55886][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263 

Please tell me what am I doing wrong ??

```
#include <Arduino.h>
#include <stdint.h>
#include <esp_camera.h>

//* for I2C
#include <Wire.h> 
TwoWire I2C_PINS_NON_DEFAULT = Wire1;   // Try to reconfig the wire Lib to use NON Default Pins

#define SCL 15                         // IO15  GPIO15  HSPI_WSO  HS2_CMD     SD-Conn
#define SDA 14                         // IO14  GPIO14  HSPI_CLK  HS2_CLK     SD-Conn

#define SETBUSSPEED                    // We do want to set the BusSpeed
#undef SETBUSSPEED                     // We do NOT want to set the BusSpeed

#ifdef SETBUSSPEED
  #define BUSSPEED 100000U             // BusSpeed = 100.000 kHz
  // #define BUSSPEED 400000U             // BusSpeed = 400.000 kHz
#endif

//* SDD1306 Config
#include <U8x8lib.h>    // I2C (Maximum Speed SCL +/- 400 kHz)                                      // ! Text Only Library
U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(/* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE); // ! Text Only (from long list)

//* PCF8574 MultiClick
#include "PCF8574.h"
#define PCF_INTERRUPT 0				         // IO0   GPIO0   CAM_PIN_XCLK or CSI_MCLK (Interrupt pin for Encoder & Switch)

// * RotaryEncoder defines (Connected to PCF8574)
#define encoderPinA  			P0
#define encoderPinB  			P1
#define switchPin					P2

// Forward declaration for the Interrupt (See Initialisation of isr_updateEncoder ISR Below), 
//   necessary here because of the Interrupt Initialisation in the next Line
void IRAM_ATTR isr_updateEncoder();	

// Initialize PCF-library for used address, SDA & SCL, Interrupt function
PCF8574 pcf8574(0x38, SDA, SCL, PCF_INTERRUPT, isr_updateEncoder);
  

void IRAM_ATTR isr_updateEncoder() {
  //! DO NOT Read pcf8574.readEncoderValue(encoderPinA, encoderPinB, &encoderValue);
  //!         and pcf8574.digitalRead(switchPin); in the isr_updateEncoder() function here !!!
  //!  this will cause "Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)"
  //!  Keep the interrupt function as short as possible and only set a flag here !!!
  //!  Next in the main loop() DO Read these values by the function getEncValues() and checkButton() !!!
  changed = true;
}

void ClrScr() {
  // ! *** See : https://github.com/olikraus/u8g2/wiki/u8x8reference
  u8x8.clear();
  delay(50);
  u8x8.clearDisplay();
  delay(50);
  u8x8.setFont(u8x8_font_chroma48medium8_r);      // Normal=1, Larger=2, largest=3, :1 pixel scale
  delay(50);                                      // See : https://github.com/olikraus/u8g2/wiki/fntlist8x8
  u8x8.print("");                                 // Draw text
  delay(50);
  u8x8.flush();
  delay(50);
}

void Display_Init() {    // ! *** Text Only
  u8x8.begin();                               // Start the Display Object, with NO PowerSaving
  delay(250);                                 // ! *** Extended delay 50 ==> 250 : See datasheet 18-12-2023 page 18 & 32
  u8x8.initDisplay();
  delay(50);         
  u8x8.setPowerSave(0);                       // Seems essentieel, for correct new start !!!
  delay(50);
  ClrScr();
}

void Display_Text(byte XPos, byte YPos, bool Invert, byte FieldSize, String MyText) {
  // u8x8 library only has LineMode (TEXT Only), for graphics use u8g2 library !!!
  // ! *** When using Graphics we are adressing X and y at pixel level !!!
  // ! *** When using Text Only we are adressing X and y at line level !!!
  // Set Font
  u8x8.setFont(u8x8_font_chroma48medium8_r);    // Small FontSize - 16 char/line
                                                // See : https://github.com/olikraus/u8g2/wiki/fntlist8x8

  // Set Normal / Inverse
  if (Invert == false) {
    u8x8.setInverseFont(0);                     // Draw 'normal' text
  } else {
    u8x8.setInverseFont(1);                     // Draw 'inverse' text
  }

  String NewStr = MyText; // Set for when FieldSize is set to 0 (NO Size Control) OR Correct !!!
  if (FieldSize != 0) {
    // Control the Length of the Text Field
    if (MyText.length() > FieldSize) {
      // MyText too long
      NewStr = "";
      for (size_t i = 1; i < FieldSize; i++) NewStr = NewStr + MyText[i];
    }

    if (MyText.length() < FieldSize) {
      // MyText too short
      NewStr = MyText;
      while (NewStr.length() < FieldSize) NewStr = NewStr + " ";
    }
  }
  u8x8.drawString(XPos, YPos, NewStr.c_str());
  u8x8.flush();
}

void setup() {
  //* Init Serial
  Serial.begin(115200);
  delay(3000);              //! We Need at least 3 seconds to start up the serial port for monitoring @ ESP32-Cam in PlatformIO !!!

  I2C_PINS_NON_DEFAULT.begin(SDA, SCL);

  //* Init OLED-Display
  Display_Init();           // Initialize the display and Start Showing the Information

  // Show texts @ the SSD Display
  mText = "Config.Finished.";
  Display_Text(0, 1, false, 16, mText);

  mText = "M.A.W.M. Nijland";
  Display_Text(0, 3, false, 16, mText);

  mText = "Apeldoorn - 2024";
  Display_Text(0, 5, false, 16, mText);

  Serial.println("\nMonitoring loop Started..");
  Serial.flush();

  //* Initialize ESP-CAM Interrupt Input
  pinMode(PCF_INTERRUPT, INPUT_PULLUP);

	//* Init PCF8574 pins	
  pcf8574.pinMode(encoderPinA, INPUT_PULLUP);   // Encoder A
  pcf8574.pinMode(encoderPinB, INPUT_PULLUP);   // Encoder B

	// Encoder pins
	pcf8574.encoder(encoderPinA, encoderPinB);

	// Encoder button
	pcf8574.pinMode(switchPin, INPUT_PULLUP);

  delay(100);

	// Start the PCF8574 library
	Serial.print("Init pcf8574...");
	if (pcf8574.begin(0x38)) {
		// Board/Chip is Acknowledging the I2C Commands
		Serial.println("OK");
	} else {
		// Board/Chip is NOT Acknowledging the I2C Commands
		Serial.println("NOT OK");
	}
}

loop() {
  // etc. etc..
}
```

xreef, 2025-01-11:
Hi,
Which device do you use? 
Bye Renzo

MAWMN, 2025-01-11:
Hi,
I am not sure about your question "Which device do you use" to what you are reffering to.
I am using ESP32-CAM AI thinker board, chip PCF8574 AN, Display SSD1306. Rotary encoder & Switch
tested the bare pcf chip with and without external pull-up resistors of 4.7kOhms to 3.3 Volt
The encoder contacts have all pull up 10kOhms to 3.3 Volts and 100nF capacitors to GND

[My test board.pdf](https://github.com/user-attachments/files/18386479/My.test.board.pdf)

thanks for looking into the issue, let me know if you need more..

MAWMN

MAWMN, 2025-01-13:
Hello, I have written the code as an PlatformIO (VsCode) project for ESP32-CAM and ESP8266 Wemos-D1-Pro board. 
Is it perhaps a good idea the upload the complete project & Schematics as a whole to investigate. 
When I use the ESP8266 (default SCL & SDA), the problems do NOT happen and all is working OKE, only when using the ESP32-CAM board (NON default SCL & SDA) the ERROR happens and can be seen in the Serial monitor after having started the blinking of the LED only, the ERROR reported is : **[457138][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263**

MAWMN

closed 2025-10-30 by xreef